Ultimaker/Cura #3569: [3.2.1] Text in right bar for Custom settings illegible.

Franks3dShop, 2018-03-27:
**Application Version**
3.2.1

**Platform**
OS X 10.13.3 on a MacBook Pro Retina Late 2013 Default resolution scaling

**Qt**

**PyQt**

**Display Driver**
  Chipset Model:	NVIDIA GeForce GT 750M
  Type:	GPU
  Bus:	PCIe
  PCIe Lane Width:	x8
  VRAM (Total):	2 GB
  Vendor:	NVIDIA (0x10de)
  Device ID:	0x0fe9
  Revision ID:	0x00a2
  ROM Revision:	3776
  Automatic Graphics Switching:	Supported
  gMux Version:	4.0.8 [3.2.8]
  Metal:	Supported, feature set macOS GPUFamily1 v3
  Displays:
Color LCD:
  Display Type:	Built-In Retina LCD
  Resolution:	2880 x 1800 Retina
  Framebuffer Depth:	24-Bit Color (ARGB8888)
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Automatically Adjust Brightness:	No

Intel Iris Pro:
  Chipset Model:	Intel Iris Pro
  Type:	GPU
  Bus:	Built-In
  VRAM (Dynamic, Max):	1536 MB
  Vendor:	Intel
  Device ID:	0x0d26
  Revision ID:	0x0008
  Automatic Graphics Switching:	Supported
  gMux Version:	4.0.8 [3.2.8]
  Metal:	Supported, feature set macOS GPUFamily1 v3

**Printer**
Any

**Steps to Reproduce**
It usually happens right when opening Cura, sometimes it is almost ok at start then it goes bad after utilisation.

**Actual Results**
See picture for reference. I the lines of text being written on screen than it seems to be overwritten by white text. Pop up help is ok.

**Expected results**
No problem in 3.0.4, fonts should be rendered correctly.

**Additional Information**
[image]
[image]
[CFG+LOG.zip](https://github.com/Ultimaker/Cura/files/1850058/CFG.LOG.zip)

Thank you.

ianpaschal, 2018-03-27:
Fixed in 3.3. Keep an eye out for the beta released today!

closed 2018-03-27 by ianpaschal

fieldOfView, 2018-03-27:
See https://github.com/Ultimaker/Cura/issues/3277
(note to @ianpaschal, this is not a Uranium bug)