Ultimaker/Cura #3570: [3.2.1] Cura crashes on startup if FDM material definitions are missing

onitake, 2018-03-27:
**Application Version**
3.2.1, but also occurs in 3.1.0.

**Platform**
Debian Linux buster amd64

**Qt**
5.9.2

**PyQt**
5.9.2

**Display Driver**
Not relevant

**Printer**
Ultimaker 3

**Steps to Reproduce**
1. Have a previous printer configuration
2. Update Cura to the latest version, without installing fdm-materials
3. Launch Cura

**Actual Results**
1. Cura crashes on startup because it can't load the FDM definitions

**Expected results**
1. Cura should launch, possibly displaying an error message

**Additional Information**
Up until 3.1, fdm-materials has been a 'soft' dependency of Cura. It was a bit less useful, but would still work. I wrote the debian/control file accordingly, making fdm-materials a "Recommends" dependency.
It seems that in 3.1.0, this changed.

I can change the behaviour, of course, but I think it would still be useful to fix the crash. What happens if a user installs a custom material definition and this definition is no longer available in a Cura update (or compatible with it)? Cura should handle this case gracefully instead of throwing.

The crash log is attached, here's just the stack trace:
```
Traceback (most recent call last):
  File "/usr/bin/cura", line 141, in <module>
    app.run()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 686, in run
    self._volume = BuildVolume.BuildVolume(root)
  File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 107, in __init__
    Application.getInstance().getMachineManager().activeQualityChanged.connect(self._onStackChanged)
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 747, in getMachineManager
    self._machine_manager = MachineManager.createMachineManager()
  File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 1328, in createMachineManager
    return MachineManager()
  File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 119, in __init__
    self.setActiveMachine(active_machine_id)
  File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 388, in setActiveMachine
    Application.getInstance().setGlobalContainerStack(containers[0])
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 160, in setGlobalContainerStack
    self.globalContainerStackChanged.emit()
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 500, in _onStackChanged
    self._updateDisallowedAreas()
  File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 607, in _updateDisallowedAreas
    disallowed_border_size = self._getEdgeDisallowedSize()
  File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 966, in _getEdgeDisallowedSize
    move_from_wall_radius = max(move_from_wall_radius, max(self._getSettingFromAllExtruders("infill_wipe_dist")))
ValueError: max() arg is an empty sequence
```

ianpaschal, 2018-03-28:
Even in 3.2 when running from source I they were treated as a soft dependency. Not sure when that changed but it's marked as a bug since what you describe as expected is how it was and should be.

Ghostkeeper, 2018-04-25:
Fixed in master now. It loads but displays the material as Empty if you have a printer with materials.

closed 2018-04-25 by Ghostkeeper

onitake, 2018-06-20:
Thank you, @Ghostkeeper !

I already made fdm-materials a required dependency since it's useful for most users, but it's good to have this crash fixed anyway.

commit 4fc29554 (2018-04-25): Don't process inheritance before there is an active global container

This could be called after an extruder was disabled by a broken configuration.

Fixes #3570.