Ultimaker/Cura #3571: [3.2.1] Print starts in the air

fatalelieberi, 2018-03-27:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10 Pro 64bit Version Version	10.0.16299 Build 16299

**Qt**
LGPLv3

**PyQt**
GPL

**Display Driver**
nvldumdx.dll Version 23.21.13.8813

**Printer**
Tevo Tornado

**Steps to Reproduce**
1. Slice the STL below
2. View the layers, at the part with many protruding walls

**Actual Results**
After moving, it picks a point where's no part below to support the print. In short, it print in the air as shown in the pictures.

**Expected results**
Cura should pick a starting point after moving where there's enough support

**Additional Information**
Results
[image]
[image]
[RoundDuctSet03.3mf.zip](https://github.com/Ultimaker/Cura/files/1851907/RoundDuctSet03.3mf.zip)

smartavionics, 2018-03-27:
The experimental bridging that will be in 3.3 avoids starting walls over air. Please give 3.3 a try and let us know if it helps. BTW, that wall looks curved, you can't print curved walls with no support underneath! Bridge lines cannot go around corners without any support.

fatalelieberi, 2018-03-27:
Okay, I just saw the 3.3 post in blog and will try it to see if it really works. 

Btw, what you said about the curved walls is true, a support is needed beneath them. However, I already tried to print using support on everything, but it generated support inside the hollow part which made removal process very difficult. Fortunately, I saw that in 3.3 there's a feature to block support which perfectly suits my need! I am very excited to test it on my print. Hopefully, it really solves my problem.

smartavionics, 2018-03-27:
Hi @fatalelieberi , my bridging code in 3.3 is dumb in that it doesn't notice when a bridge skin area is actually over support (and so should not be considered a bridge). I have rectified this but my changes have not yet made it into a release.

Ghostkeeper, 2018-04-24:
You really need that to be hollow, that's going to be a problem for FDM printing in general...

Starting from a different point is going to collide with our current code of where it should start for aligning the Z seam and the order in which walls are being printed...

diegopradogesto, 2019-09-11:
I cannot reproduce this issue with the latest Cura version and the default profiles for the Tevo Tornado. Maybe it's a different profile with different settings. In case it's still happening, please provide a project file using the latest stable version of Cura.

I will close this in the meantime, but feel free to reopen if necessary.

closed 2019-09-11 by diegopradogesto