Ultimaker/Cura #3572: Cura Connect - Print marked finished while printing.

Spirou42, 2018-03-27:
Hi i don't know if this is the right repo but while i haven't succeed in finding a better one i'll dump my findings here:
**Application Version**
4.3.2.20180308

**Platform**
Ultimaker 3

**Printer**
Same as Platform

**Steps to Reproduce**
Start a print from Cura and open the web view of the printer the file was sanded to.

**Actual Results**
Print is marked as finished, 
[image]

[image]

**Expected results**
I would expect that the message indicates the actual state the printer is in.

**Additional Information**
Slicer: Cura 3.2.1 using adaptive layer height.

ianpaschal, 2018-03-28:
You're in the right place. We tend to try and funnel all issues here since GitHub doesn't have an easy mechanism for moving issues to a different repository. And congratulations, you're one of the 0.1% of people who actually put effort into that. Thank you, haha.

Also, paging @nallath

Ghostkeeper, 2018-04-24:
Doesn't reproduce for me, even when I disconnect from the network while printing or something.

Spirou42, 2018-04-25:
Ok, so how can I help to get the needed information?

ianpaschal, 2018-04-25:
Can you share the Cura project (just add .zip to the end of the file name since it's technically a zip file already)?

Then we can try to recreate the scenario _exactly_.

If we're still not able to reproduce it then I'm afraid we can't really try to fix it, but maybe we will see that there's something specific in your setting combination or model that is triggering it.

Spirou42, 2018-04-26:
I can reproduce it  every time. The discrepancy between the display in Cura and in Cura Connect are consistent. 
[DummyTest.3mf.zip](https://github.com/Ultimaker/Cura/files/1950918/DummyTest.3mf.zip)
Created the project for testing a few minutes ago. 

CuraConnect in Safari:
[image]

Cura (3.3.1) display of printer status at almost the same time:
[image]

Spirou42, 2018-04-26:
Is it possible, that some print job status entry got stuck on the UM3 so it got displayed in CuraConnect over and over again? I ask that question cause the "finished" entry does not display the name of the last print but always the same "old" name. The status of the printers cores on the other hand is updated and showed the correct values.
So maybe there is an status snippet in CuraConnect on my printer, that the software can't get rid off cause there is no "real" print job any more that can trigger a state change for this orphan status.

Spirou42, 2018-05-24:
Oops, there might be a slight misunderstanding here. 
The actual print starts, runs, and finnisches without any problem. The problem I have described seams to be a "display only" problem. 
The name of the print marked as finished stays the same. It is an old print job name. The current print job works well but is not displayed in CuraConnect correctly. Thats why i asked the question about a possibly stuck job status on my UM3 a while ago (directly above your comment).

Spirou42, 2018-06-23:
after resetting cura connect on the printer, the problem has gone.

closed 2018-06-23 by Spirou42