Ultimaker/Cura #3573: Cura Connect  4.3.2.20180308 video stream persist while video window is closed.

Spirou42, 2018-03-27:
Another issue for Cura Connect posted here cause i have not found a better place.
**Application Version**
4.3.2.20180308

**Platform**
Ultimaker 3

**Printer**
same as platform

**Steps to Reproduce**
Browsers:  Chrom, Safari and FireFox
Open your favourite network monitor and filter for your printers IP address.
Open cura connect on your printer and open any video window. 
You should see the transfer from CC to your browser in your network monitor.
close the video feed in the browser and notice, that the transfer volume does not drop to normal again.

**Actual Results**
The video feed is streamed to the browser even when the video window is not visible.

**Expected results**
The network transfer should stop at the moment the video window is invisible.

**Additional Information**
Have tested this on: macOS 10.13.3, 
Network Filter/Monitor: Little Snitch Version 4.0.6 (5124)
Safari 11.0.3 
Chrome 65.0.3325.181 
FireFox: 59.0.2 (64-bit)
all three browsers showed the same behaviour.

ianpaschal, 2018-03-28:
Paging @nallath

Ghostkeeper, 2018-04-25:
This is now fixed in the next firmware update. Thank you!

closed 2018-04-25 by ianpaschal