Ultimaker/Cura #3574: [3.3.0 Beta] Creating New Material Causes Crash

fatalelieberi, 2018-03-27:
**Application Version**
Cura 3.3.0 Beta

**Platform**
Windows 10 Pro 64bit Version Version	10.0.16299 Build 16299

**Qt**
LGPLv3

**PyQt**
GPL

**Display Driver**
nvldumdx.dll Version 23.21.13.8813

**Printer**
Tevo Tornado

**Steps to Reproduce**
1. Open Manage Material dialog box
2. Click on Create button

**Actual Results**
Cura just closes without any warning

**Expected results**
New material can be created

**Additional Information**
[cura.log.txt](https://github.com/Ultimaker/Cura/files/1852637/cura.log.txt)

jackha, 2018-03-29:
Weird, the log didn't register the crash / stacktrace. With a clean install it seems fine.

Did you use an older version of Cura? In that case the configuration was upgraded and maybe something in there is causing this. If so, can you provide us with the Cura configuration folder?

fatalelieberi, 2018-03-29:
Yes, I did use 3.2.1 and I hadn't uninstalled it when I installed the 3.3, but the issue still happens even after I uninstalled the 3.2.1. You can see the content of my configuration folder here.

[Cura 3.3 Configuration Files.zip](https://github.com/Ultimaker/Cura/files/1859531/Cura.3.3.Configuration.Files.zip)

fatalelieberi, 2018-03-29:
Tried clean install in my computer, still the same, crashes whenever Create button is pressed.

jackha, 2018-03-29:
Thanks for sharing the configuration folder. 

To solve it, you can clean the config folders. Delete all the contents of the 3.3 folder (from Cura: Help -> Show configuration folder), but leave the 3.3 folder itself. Then restart and try again.
Cura does nothing with the (windows)registry and such, so emptying the config folder is actually really starting clean. Uninstalling Cura actually leaves the config folders, which is currently a known issue.

fatalelieberi, 2018-03-29:
So, basically I did what you told me. I deleted all files in the folder (which made me lost all of my settings & profiles T.T) and after I opened Cura, I was asked to configure my printer again. However, now the crash has moved from creating materials to creating a printer. In the log, you can see there are some lines indicating some errors. I hope you can fix this.

[cura.log](https://github.com/Ultimaker/Cura/files/1860607/cura.log)

fatalelieberi, 2018-04-08:
A little update, after I deleted the whole 3.3 folder, the Cura could run again and it imported all configurations from 3.2 folder. However, the issue with material creation still exists.

Ghostkeeper, 2018-04-24:
It does log the error, at least in the last log. This is the relevant bit:

    2018-03-29 21:47:02,397 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
    2018-03-29 21:47:02,399 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
    2018-03-29 21:47:02,400 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 354, in addMachine
    2018-03-29 21:47:02,402 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 74, in createMachine
    2018-03-29 21:47:02,403 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Machines\MaterialManager.py", line 396, in getDefaultMaterial
    2018-03-29 21:47:02,405 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Machines\MaterialManager.py", line 236, in getRootMaterialIDForDiameter
    2018-03-29 21:47:02,406 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'NoneType' object has no attribute 'get'

Ghostkeeper, 2018-04-24:
Ah it seems this was fixed yesterday: https://github.com/Ultimaker/Cura/commit/f62ce1966b50c5046a1b4c34e4282a4488da574c

closed 2018-04-24 by Ghostkeeper