Ultimaker/Cura #3575: 3.3.0 speed settings need enter key hit for setting to take effect

scrat82, 2018-03-27:
**Application Version**
 beta 3.3.0 beta

**Platform**
 ultimaker 2+

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2018-03-27:
Hi, please try entering the speed digits slowly (around 1 per second) followed by enter or tab. If you do that and the old setting does not reappear, then it's the same issue as I am seeing (#3564).

scrat82, 2018-03-27:
I have waiting for it to slice and watch other settings go back to a different value. If I highlight the value to change instead of clicking on the box and using the delete key I get different results too.

smartavionics, 2018-03-27:
Hmm, it appears to have a problem... LOL, I expect someone will wake up and fix it before too long!

scrat82, 2018-03-27:
I agree. I was looking at the layer view, and when I highlighted the infill and changed that, the layer view changed instead. Almost if I read the mouse to far to the left when I'm highlighting the number, it jumped over to the layer height box in the view window

darkdvd, 2018-03-27:
Same problem for me (CURA 3.2.1 / 3.3)
Windows 10 x64
CURA in "compatibility mode"

jackha, 2018-03-29:
Really weird, I see it too on my UM3 in linux.
Devs: CURA-5176

macaron16, 2018-04-01:
same issue on osX.
If we click 'tab key' no effect (the value don't change) 
Not only for speed setting, but for all settings.

smartavionics, 2018-04-01:
Yes, the UI appears to be well broken in 3.3 beta. There's this problem with entering values into settings and also the problem with editing start/end gcode. I am mildly shocked that it was released in such a broken state.

lolorc, 2018-04-11:
confirmed, it's very annoying

lolorc, 2018-04-12:
@smartavionics oh that's good news, i'll build master this evening then.
what's the commit ? looks like I missed it.

smartavionics, 2018-04-12:
Sorry, I forget now, I think it was in Uranium (maybe 7b53e35fcb7f28ca4cda43a79102f5bfee44106d).

lolorc, 2018-04-12:
looks like it, a revert, thanks

diegopradogesto, 2018-04-13:
This was fixed on April 6th and it will be in 3.3.

closed 2018-04-13 by diegopradogesto