Ultimaker/Cura #3576: Cura 3.3B crashes on first startup while parsing custom variant file

Spirou42, 2018-03-27:
**Application Version**
official 3.3b

**Platform**
macOS 10.13.3

**Display Driver**
AMD FirePro D700 6 GB

**Printer**
Ultimaker 3

**Steps to Reproduce**
Have a custom variant based on ultimaker3_aa04.inst.cfg from cura 3.2.1 which is missing the hardware_type entry in the [metadata] section.
start cura 3.3b for the first time (3.3 directory in the cura configuration directory)

**Actual Results**
crash 

**Expected results**
ignore the  "formerly correct" file, start up and inform the user, that there is a problem with one of the custom configuration files.

**Additional Information**

ianpaschal, 2018-03-28:
I have a strong suspicion of the problem, but can you share your custom variant file so we can check for sure?

Spirou42, 2018-03-28:
Jep no problem, also have attached them to the crash report.
[CustomVariantFiles.zip](https://github.com/Ultimaker/Cura/files/1855477/CustomVariantFiles.zip)
The crash disappears, if i add *hardware_type* = *nozzle* in the [metadata] section 

btw, what other hardware_types are valid entries? I only have found he value *nozzle* inside the applications bundle.

jackha, 2018-03-29:
Yes this looks like a problem of the setting_version, it should have been increased. 
Devs: CURA-5175

closed 2018-03-29 by diegopradogesto

Spirou42, 2018-03-29:
Thanks for the quick answer Diego.