Ultimaker/Cura #3577: Cura 3.3B- First impressions

Thisismydigitalself, 2018-03-27:
3.3B First Impressions:

- Only Half of my profiles were copied to 3.3.
- Printer's start/end codes are completely different
- Printer's settings are different.
- LineType view much more "noisy" than in 3.2.1

ianpaschal, 2018-03-28:
We really appreciate your feedback, especially since you're an active member around here.

But, that being said, could you please open actual issues about these things with examples and detailed info? Otherwise there's not really anything we can do with the information.

I'm going to close this because it's too hard to use the issue system when it's several different topics/issues all mashed into one.

closed 2018-03-28 by ianpaschal

Thisismydigitalself, 2018-03-28:
sure thing. its just too many issues surfaced wth 3.3b including crash which i havnt included so i just dumpped my first impressions. as i am on the road i will try to manage issues the proper way soon.