Ultimaker/Cura #3580: Cura 3.3 crashes when trying to access the Plugin Browser

MediaXYZ, 2018-03-28:
Hi There, 

Forgive me if I've screwed up this bug submission. I've never used GitHub before.

Cura 3.3 Beta seems to be crashing on my MacBook Air whenever I try to open the Plugin Browser. It seems to be repeatable issue that has persisted across a system restart.

**Application Version**
3.3 Beta

**Platform**
Mac OS 10.13.3 (High Sierra)

**Qt**
Beta version from Ultimaker's website

**PyQt**
Beta version from Ultimaker's website

**Display Driver**
Intel HD Graphics 5000

**Printer**
Custom Printer

**Steps to Reproduce**
Select "Browse Plugins..." from the "Plugins" menu

**Actual Results**
The entire application aborts and closes. The Apple "Ultimaker Cura quit unexpectedly" pop up appears prompting to send crash data to Apple.

**Expected results**
The plugin browser should have opened

**Additional Information**

Here is some of the top level stuff from the crash report:

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x10:
--> 
    __TEXT                 000000010e5d0000-000000010e7a1000 [ 1860K] r-x/rwx SM=COW  * [/Applications/Ultimaker Cura.app/Contents/MacOS/cura]

Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x00007fff7e6ede3e __pthread_kill + 10

MediaXYZ, 2018-03-28:
Turns out exactly the same crash with the same exceptions occurs if I try to add my API from my OctoPrint to my printer (as I paste it in to the box). This is also repeatable.

Vandrasc, 2018-03-28:
@fieldOfView , is this related to the upgrade of the Octo plugin? Thank you!

fieldOfView, 2018-03-28:
Nope, though the crash has the same cause. It is the QNetworkAccessManager in Qt that also crashes Macs when adding an UM3, see #2617 and #2622

MediaXYZ, 2018-03-29:
Although I use the OctoPrint plugin, I think it is an broader issue than that.

ChrisTerBeke, 2018-03-29:
The plugin browser also does network requests, so it's probably an issue with the Qt networking on this specific configuration of OS version and Qt version.

CubeFactory2, 2018-04-24:
Same here, Cura crashes before the program start has finished. Went back to 3.2. ;o(
macOS High Sierra 10.13.4
UM3 via network

CubeFactory2, 2018-04-24:
Same here,

Cura crashes before the program start has finished. 
Went back to 3.2 ;o(

macOS High Sierra 10.13.4
UM 3 via network

Ghostkeeper, 2018-05-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/2617.

closed 2018-05-11 by Ghostkeeper

LipuFei, 2018-06-12:
Hi all, we have created a 3.4 BETA build for mac with Qt 5.10 hoping it can fix this issue. Could you try it out see if problem still exists? Thank you all!

Download link: https://we.tl/43i1hChi8m
In case the link expires, let us know so we can upload a new one.

LipuFei, 2018-06-12:
Sorry guys... I only uploaded the linux AppImage, this is the link which contains all installers/AppImage/DMG.
https://we.tl/MLmTYeGuW5

MediaXYZ, 2018-06-12:
Hi there,

Nope - the application still crashes when I try to browse plugins in what is now called the “toolbox” in the 3.4 beta  :(

Regards,

Luke

https://we.tl/MLmTYeGuW5<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwe.tl%2FMLmTYeGuW5&data=02%7C01%7C%7C5a514f6ff8b24f920bb508d5d040fce6%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636643899886182791&sdata=y7OJcZlaXp8se90cEA6i449yhjh1DTcdhkLJGhdcX60%3D&reserved=0>

MediaXYZ, 2018-06-12:
Here is the crash log:

Process:               cura [22150]
Path:                  /private/var/folders/*/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               3.4 (3.4.0-internal-master.20180611011342)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [22150]
User ID:               502

Date/Time:             2018-06-12 20:43:54.834 +1000
OS Version:            Mac OS X 10.13.4 (17E202)
Report Version:        12
Anonymous UUID:        BAEB0D1F-0A9A-57F9-4D05-7FDC46F79D46

Sleep/Wake UUID:       864E7FCA-6F26-4771-ACF5-4387B1F23905

Time Awake Since Boot: 150000 seconds
Time Since Wake:       5000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x10:
-->
    __TEXT                 000000010f61f000-000000010f7f0000 [ 1860K] r-x/rwx SM=COW    [/var/folders/xm/9yrrv4d55cx8s_h0qgwfz78r0000gp/T/AppTranslocation/6C39CB93-BFC4-4097-A698-FBBC734047DA/d/Ultimaker Cura.app/Contents/MacOS/cura]

Thread 0 Crashed:
0   libsystem_kernel.dylib         0x00007fff71754b6e __pthread_kill + 10
1   libsystem_c.dylib             0x00007fff716626fe raise + 26
2   libsystem_platform.dylib       0x00007fff71912f5a _sigtramp + 26
3   ???                           000000000000000000 0 + 0
4   com.apple.CoreFoundation       0x00007fff49308a61 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
5   com.apple.CoreFoundation       0x00007fff493c247c __CFRunLoopDoSource0 + 108
6   com.apple.CoreFoundation       0x00007fff492eb4c0 __CFRunLoopDoSources0 + 208
7   com.apple.CoreFoundation       0x00007fff492ea93d __CFRunLoopRun + 1293
8   com.apple.CoreFoundation       0x00007fff492ea1a3 CFRunLoopRunSpecific + 483
9   libQt5Network.5.dylib         0x0000000110dc8d75 0x110d24000 + 675189
10  libQt5Network.5.dylib         0x0000000110dc9b51 QNetworkProxyFactory::systemProxyForQuery(QNetworkProxyQuery const&) + 17
11  libQt5Network.5.dylib         0x0000000110dbf753 0x110d24000 + 636755
12  libQt5Network.5.dylib         0x0000000110dc1842 QNetworkProxyFactory::proxyForQuery(QNetworkProxyQuery const&) + 50
13  libQt5Network.5.dylib         0x0000000110d5555e 0x110d24000 + 202078
14  libQt5Network.5.dylib         0x0000000110d73329 0x110d24000 + 324393
15  libQt5Network.5.dylib         0x0000000110d78283 0x110d24000 + 344707
16  libQt5Network.5.dylib         0x0000000110d6fdee 0x110d24000 + 310766
17  libQt5Network.5.dylib         0x0000000110d6fa9b 0x110d24000 + 309915
18  libQt5Network.5.dylib         0x0000000110d53ed5 QNetworkAccessManager::createRequest(QNetworkAccessManager::Operation, QNetworkRequest const&, QIODevice*) + 1653
19  PyQt5.QtNetwork.so             0x0000000110c80ef1 sipQNetworkAccessManager::createRequest(QNetworkAccessManager::Operation, QNetworkRequest const&, QIODevice*) + 129
20  libQt5Network.5.dylib         0x0000000110d522de QNetworkAccessManager::get(QNetworkRequest const&) + 30
21  PyQt5.QtNetwork.so             0x0000000110c81d01 meth_QNetworkAccessManager_get(_object*, _object*) + 113
22  nl.ultimaker.cura             0x000000010f674095 PyCFunction_Call + 165
23  nl.ultimaker.cura             0x000000010f6f14c0 PyEval_EvalFrameEx + 25584
24  nl.ultimaker.cura             0x000000010f6f64a2 0x10f61f000 + 881826
25  nl.ultimaker.cura             0x000000010f6f0e0f PyEval_EvalFrameEx + 23871
26  nl.ultimaker.cura             0x000000010f6f64a2 0x10f61f000 + 881826
27  nl.ultimaker.cura             0x000000010f6f0e0f PyEval_EvalFrameEx + 23871
28  nl.ultimaker.cura             0x000000010f6f5cb2 0x10f61f000 + 879794
29  nl.ultimaker.cura             0x000000010f6eb0ae PyEval_EvalCodeEx + 78
30  nl.ultimaker.cura             0x000000010f65462d 0x10f61f000 + 218669
31  nl.ultimaker.cura             0x000000010f62af75 PyObject_Call + 101
32  nl.ultimaker.cura             0x000000010f640cbd 0x10f61f000 + 138429
33  nl.ultimaker.cura             0x000000010f62af75 PyObject_Call + 101
[... 626 lines cut ...]
       0x119779000 -        0x1197cbff7 +libqtiff.dylib (0) <AD090D18-9AC8-3BD2-A90D-BB23700775A0> /var/folders/*/Ultimaker Cura.app/Contents/Resources/imageformats/libqtiff.dylib
       0x1197d3000 -        0x1197d7fff +libqwbmp.dylib (0) <3F960D3F-EE67-388A-B412-2F1BB4018090> /var/folders/*/Ultimaker Cura.app/Contents/Resources/imageformats/libqwbmp.dylib
       0x1197db000 -        0x119859ffb +libqwebp.dylib (0) <BB37FCAD-4A0F-3CA8-A144-DC2455367415> /var/folders/*/Ultimaker Cura.app/Contents/Resources/imageformats/libqwebp.dylib
       0x119a91000 -        0x119a99ff7 +_elementtree.cpython-35-darwin.so (???) <ACFDC2B6-FA70-3A9A-B757-5A69C3572641> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/_elementtree.cpython-35-darwin.so
       0x119aa2000 -        0x119aabff7 +_socket.cpython-35-darwin.so (???) <E8AD149E-4DBF-374C-BEF9-8D482F88E58D> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/_socket.cpython-35-darwin.so
       0x119ab7000 -        0x119ab7fff +_scproxy.cpython-35-darwin.so (???) <208A1C7C-35DA-36BD-9A2D-2D2C2C687D5E> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/_scproxy.cpython-35-darwin.so
[... cut ...]
       0x61d3f3000 -        0x61d452ff7 +libQt5DBus.5.dylib (0) <D2C63382-3452-3EDE-9367-CCF21433243F> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/libQt5DBus.5.dylib
       0x61d46e000 -        0x61d499ff7 +libQt5Xml.5.dylib (0) <7685EB6B-87CF-3BC8-B188-79EC00039597> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/libQt5Xml.5.dylib
       0x61d4ad000 -        0x61d4e1ff7 +libQt5OpenGL.5.dylib (0) <E8BC6A96-4DB6-3DA5-9FD6-79FE2BE098E4> /var/folders/*/Ultimaker Cura.app/Contents/MacOS/libQt5OpenGL.5.dylib
       0x61d505000 -        0x61d52eff7 +PyQt5.QtSql.so (0) <5BBDCE0E-D6B6-3CC7-B82A-481B2666C144> /var/folders/*/Ult

LipuFei, 2018-06-12:
Hi @MediaXYZ , I think I made a mistake uploading a wrong installer again (too many installers in my folder)... Sorry about that. Could you try this one? thank you!

https://we.tl/gu0kqipSX6

MediaXYZ, 2018-06-15:
So this version seems to work OK.

I can no longer replicate the previous errors, and was able to both open the plugin browser, install plugins, and also  an API for OctoPrint. The rest of the software seems to behave normally, and I haven’t encountered any new errors yet.

Great!

https://we.tl/gu0kqipSX6<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwe.tl%2Fgu0kqipSX6&data=02%7C01%7C%7Ce5266cca6d34474d886708d5d059bb3b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636644006164611282&sdata=5jeoL0aio6af3VtUZcyMi3GWiMou6KfoMLXDMvom4d4%3D&reserved=0>

noamjo, 2018-07-03:
@ LipuFei , im now experiencing the same problem with Cura 3.4 , running macOS Sierra 10.12.6
Is there anyway you can share a working version ?
Already tried uninstalling the older software, rebooting the computer, etc.

LipuFei, 2018-07-03:
Hi @noamjo , here is a new WeTransfer link for it: https://we.tl/6VbdYHoGLN

amandreamartino, 2018-07-15:
Could you plese share the link again? I'm having the exact same issue

LipuFei, 2018-07-16:
Hi @amandreamartino , here is a 3.4.1 build with Qt 5.10. It will expire in 7 days.

https://we.tl/wc8OGJ6BgP

Jasonpgr44, 2018-07-18:
I've been having this issue when cura tries to do anything network related. I tried the newest version from the ultimaker site with no luck, but the version you linked to fixed the issue perfectly. Thank you!

LipuFei, 2018-07-18:
Starting from the next release, we will use Qt 5.10, so this problem will be fixed.

jwveldhuis, 2018-08-03:
@LipuFei I think the next release is not yet available? Can I get a download link to 3.4.1 with Qt 5.10? Same issue here with Mac 10.13.6

LipuFei, 2018-08-03:
@jwveldhuis Here you go **https://we.tl/QsHpY4xRYT**

micaheli, 2018-09-09:
I need an updated version to this for 3.4.1.  Mine is still crashing.

fieldOfView, 2018-09-09:
Word on the street is there will be a new beta soon

Ghostkeeper, 2018-11-09:
Download a link here: https://github.com/Ultimaker/Cura/releases

Since 3.5.0 we have the new Qt version, so if you can test that it starts properly that would help.