Ultimaker/Cura #3582: Slicer Engine will not run.

Macflame, 2018-03-28:
**Application Version**
<!Wanhao 1.1 Beta-- -->

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Vandrasc, 2018-03-28:
Hi @Macflame, could you provide more info? How can we reproduce it? Also helpful if you will share your logs with us, thanks!

Macflame, 2018-03-28:
Hi Vandrasc,
Many thanks for your swift response.
I am currently using Windows 10 Home (Microsoft Insider version 17127 rs4 release). 
I have reinstalled the Wanhao Cura Beta 1.1 program and 'sliced' two models.
I closed down the program, then reopened it to prepare another print. The slicer will not work and just hangs.
I can exit the slicer and use the other parts of the program.
I attach my zipped log and configuration files for your information.
I hope this helps.
[LogCFG.zip](https://github.com/Ultimaker/Cura/files/1856633/LogCFG.zip)
Patrick.

closed 2018-03-28 by Macflame

fieldOfView, 2018-03-28:
I don’t think we can do much to support this version of Cura, please refer to Wanhao support. We don’t know what they changed in their version, or what version of Cura they based theirs on. And if we could fix it, we could not fox it in the Wanhao version.

Macflame, 2018-03-28:
Many thanks. I shall have a chat with Wanhao.