Ultimaker/Cura #3583: Cura 3.1.0 Seems to get stuck when open attached STL

cady, 2018-03-28:
**Application Version**
3.1.0

**Platform**

Windows 10 
**Qt**

**PyQt**

**Display Driver**

NVidia 391.35 (GTX 970)
**Printer**

Not relevant, I think. But I can figure out how to get it if it's really necessary. Cura won't open
**Steps to Reproduce**

Open attached STL
**Actual Results**

**Expected results**

Cura seems to get stuck and crash
**Additional Information**

2018-03-28 23:17:41,520 - DEBUG - STLReader.STLReader.read [79]: Loaded a mesh with 75000 vertices
2018-03-28 23:17:41,544 - WARNING - UM.Mesh.MeshData.approximateConvexHull [368]: Stopped shrinking data because otherwise the convex hull calculation will crash. Vertices: 75000 (target: 51200)
2018-03-28 23:17:41,598 - ERROR - UM.Logger.logException [76]: Exception: Exception occurred while loading file C:/Users/Cady/Downloads/AA_-_USB_Battery_Pack.stl
2018-03-28 23:17:41,600 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2018-03-28 23:17:41,601 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\FileHandler\ReadFileJob.py", line 66, in run
2018-03-28 23:17:41,602 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshFileHandler.py", line 37, in readerRead
2018-03-28 23:17:41,604 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 180, in getExtents
2018-03-28 23:17:41,605 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 250, in getConvexHullVertices
2018-03-28 23:17:41,606 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 242, in getConvexHull
2018-03-28 23:17:41,608 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 234, in _computeConvexHull
2018-03-28 23:17:41,609 - ERROR - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 381, in approximateConvexHull
2018-03-28 23:17:41,610 - ERROR - UM.Logger.logException [80]:   File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
2018-03-28 23:17:41,611 - ERROR - UM.Logger.logException [80]:   File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
2018-03-28 23:17:41,612 - ERROR - UM.Logger.logException [80]: scipy.spatial.qhull.QhullError: QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate
2018-03-28 23:17:41,614 - ERROR - UM.Logger.logException [80]: 
2018-03-28 23:17:41,615 - ERROR - UM.Logger.logException [80]: While executing:  | qhull i Qt
2018-03-28 23:17:41,616 - ERROR - UM.Logger.logException [80]: Options selected for Qhull 2015.2.r 2016/01/18:
2018-03-28 23:17:41,617 - ERROR - UM.Logger.logException [80]:   run-id 1789975916  incidence  Qtriangulate  _pre-merge  _zero-centrum
2018-03-28 23:17:41,618 - ERROR - UM.Logger.logException [80]:   _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
2018-03-28 23:17:41,620 - ERROR - UM.Logger.logException [80]:   Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0
2018-03-28 23:17:44,611 - WARNING - UM.Mesh.MeshData.approximateConvexHull [368]: Stopped shrinking data because otherwise the convex hull calculation will crash. Vertices: 75000 (target: 51200)
2018-03-28 23:17:44,663 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-03-28 23:17:44,664 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-03-28 23:17:44,665 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 770, in event
2018-03-28 23:17:44,667 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 297, in event
2018-03-28 23:17:44,668 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189, in call
2018-03-28 23:17:44,669 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
2018-03-28 23:17:44,670 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Scene\SceneNode.py", line 136, in calculateBoundingBoxMesh
2018-03-28 23:17:44,671 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Scene\SceneNode.py", line 616, in getBoundingBox
2018-03-28 23:17:44,672 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Scene\SceneNode.py", line 680, in _calculateAABB
2018-03-28 23:17:44,674 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 180, in getExtents
2018-03-28 23:17:44,675 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 250, in getConvexHullVertices
2018-03-28 23:17:44,676 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 242, in getConvexHull
2018-03-28 23:17:44,677 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 234, in _computeConvexHull
2018-03-28 23:17:44,678 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 381, in approximateConvexHull
2018-03-28 23:17:44,679 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
2018-03-28 23:17:44,681 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
2018-03-28 23:17:44,682 - CRITICAL - cura.CrashHandler.__init__ [65]: scipy.spatial.qhull.QhullError: QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate
2018-03-28 23:17:44,683 - CRITICAL - cura.CrashHandler.__init__ [65]: 
2018-03-28 23:17:44,684 - CRITICAL - cura.CrashHandler.__init__ [65]: While executing:  | qhull i Qt
2018-03-28 23:17:44,685 - CRITICAL - cura.CrashHandler.__init__ [65]: Options selected for Qhull 2015.2.r 2016/01/18:
2018-03-28 23:17:44,686 - CRITICAL - cura.CrashHandler.__init__ [65]:   run-id 1790026337  incidence  Qtriangulate  _pre-merge  _zero-centrum
2018-03-28 23:17:44,688 - CRITICAL - cura.CrashHandler.__init__ [65]:   _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
2018-03-28 23:17:44,689 - CRITICAL - cura.CrashHandler.__init__ [65]:   Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0
[stderr.log](https://github.com/Ultimaker/Cura/files/1857617/stderr.log)

ianpaschal, 2018-03-29:
This slices fine in 3.3 beta. Please try upgrading as we likely solved the problem you're encountering.

BagelOrb, 2018-04-03:
Also please attach .curaproject.3mf file rather than stl file in the future.

Ghostkeeper, 2018-04-26:
The Cura project file won't make any difference since this is an error in QHull that happens when we compute the convex hull for the collision area of the object. STL is fine.

But indeed, I believe this is fixed in 3.3. When loading I get a message that it scaled the model to 100000%. Previously it was probably computing the convex hull before scaling and then ran into precision issues.

closed 2018-04-26 by Ghostkeeper