Ultimaker/Cura #3588: Cura 3.3B profiles - Imported half of my 3.2.1 Profiles.

Thisismydigitalself, 2018-03-29:
**Application Version**
3.3B
**Platform**
Win10/64
 
**Display Driver**
NV 391.01
 
**Steps to Reproduce**
Just install 3.3B and check Profiles list (can't take screenshot as Cura UI closes this pull down menu when trying to capture).

**Actual Results**
Expect to see all my profiles imported/copied over to 3.3
**Expected results**
About half my profiles are not shown.

jackha, 2018-03-29:
Can you share your 3.2.1 config folder? That would really help us debug this.

mtmello2003, 2018-03-29:
Same issue here....

Example:
[image]

diegopradogesto, 2018-05-09:
I will close this since it seems to be the same as #3589

closed 2018-05-09 by diegopradogesto