Ultimaker/Cura #3589: Cura 3.3 Printer settings (Custom FDM)

Thisismydigitalself, 2018-03-29:
**Application Version**
3.3B
**Platform**
Win10/64
 
**Display Driver**
NV 391.01
 
**Steps to Reproduce**
after installing 3.3B I checked my printer settings. they here all wrong - Start/End gcodes and also the printer profile as well.

**Actual Results**
Wrong settings (same printer name though..)
**Expected results**
Expect to see all my printers imported/copied over to 3.3 as-is in 3.2.1

jackha, 2018-03-29:
I think you upgraded from 3.2.1? Is it possible to get your 3.2.1 config folder?
What's wrong? Does it give different values / is it empty / something else?

diegopradogesto, 2018-05-09:
@Thisismydigitalself can you try with 3.3.1 if it still happens? If not, can you please send us the 3.2 data folder?

ianpaschal, 2018-05-28:
Never heard back; closing.

closed 2018-05-28 by ianpaschal