Ultimaker/Cura #3590: Cura 3.3B - Settings values sometimes (not always) revert back after slicing

Thisismydigitalself, 2018-03-29:
Application Version
3.3B
Platform
Win10/64

Display Driver
NV 391.01

Steps to Reproduce
install 3.3B, change setting parameter and slice - I noticed that when slicing finishes layer height changes back to what it was before the change.

Actual Results
settings values changes without user intervention.

Expected results
Obviously Setting should keep their given values

jackha, 2018-03-29:
That's really weird... obviously it doesn't happen on my computer... so I want to try to get some solid reproduction steps.
- Does it always do that?
- Any "special" settings / machines / profiles?
- Is the setting in focus / out of focus / did you use enter / tab / something else?

alekseisasin, 2018-03-29:
@mtmello2003 , I have reproduced this issue. We will work on it.
Devs: CURA-5181

smartavionics, 2018-03-29:
Hi folks. For me, when I changed a value it would revert back if I entered the value using my normal typing speed but if I slowed down and entered, say, 1 digit per second then the new value would be accepted. Does the same occur for you?

joyfromprocess, 2018-03-29:
@smartavionics - I'm seeing the same behavior.

If I change a number using my normal typing speed and then tab to next field, some fields reset to the previous number. I can't make it happen in every field, but I can still make it happen a lot.

If I force myself to slow down, the field keeps my value. It's easy to workaround but extremely frustrating when I forget and try to work at my normal speed, because I invariably realize I need to jump back and fix multiple fields.

smartavionics, 2018-03-29:
Ha! thanks @joyfromprocess , at least that means it's not just me going mad!

bcappel, 2018-04-01:
This happens to me as well. And if I slow down my typing it finally takes.

smartavionics, 2018-04-06:
Hi @jackha, is anyone working on a fix for this problem because it's a real pain. I have also noticed that changes to the start/end gcode and also entering text into the postprocessor plugin fields is similarly broken. Perhaps all these problems are related?

fieldOfView, 2018-04-06:
Might have just been fixed here: https://github.com/Ultimaker/Cura/issues/3338#issuecomment-379177982

smartavionics, 2018-04-06:
Yes, I think that change has fixed the problem of setting changes being ignored unless they are typed at snail pace!

Ghostkeeper, 2018-04-06:
Ah yes, this is indeed the issue that we fixed today. This Github issue is even linked to the Jira ticket.

closed 2018-04-06 by Ghostkeeper