Ultimaker/Cura #3592: 3MF about multi-material

SWAssassin, 2018-03-29:
Hello，
I am making a multi-material 3D printer that uses Stereo lithography Appearance. I want to know if 3MF includes more than one material information. The CURA shows only one material.  Can I get a multi-material slice through CURA for secondary development?
THANK YOU !

alekseisasin, 2018-03-29:
Hi @SWAssassin , Answering to your question, Cura shows only one material. In Cura after loading we do not read any object properties from the meshe. After loading Cura applies Extruder settings which comes together with Cura. 
If you have proposal or solution for this feature you can create a Pull Request where you describe its functionality. Together with other people we might come to nice solution and this feature will be added to Cura.

ChrisTerBeke, 2018-03-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/2539.

closed 2018-03-29 by ChrisTerBeke