Ultimaker/Cura #3594: Improvement Request: Adhesion extruder option

mtmello2003, 2018-03-29:
Please add a option to "no override" to build plate adhesion > build plate adhesion extruder that will use the active extruder.

Actually, I need to choose between left or right (2 extruder in my case).

[image]

ChrisTerBeke, 2018-03-29:
There is no default for this as the setting is not inherited, so you can't 'not override'.

mtmello2003, 2018-03-29:
in 3.3B there is an option to disable and make active an extruder btw I guess if I use select just one extruder with a part to be printed, this one is the "default".

BagelOrb, 2018-04-04:
It is not clear to me whether your suggestion is what the OP means.

Could you explain the issue a bit more? Could you formulate the problem you are faced with without the suggested feature?

Ghostkeeper, 2018-04-26:
We've solved this in a different way: If you disable an extruder but had one of these settings (such as Build Plate Adhesion Extruder) set on the extruder that you disabled, it'll select a different extruder that is still active.

So now if you disable all extruders except one, the extruder setting will always end up on the one extruder that you didn't disable.

closed 2018-04-26 by Ghostkeeper

mtmello2003, 2018-04-26:
Perfect solution! tks