Ultimaker/Cura #3596: [Cura 3.2.1] When I click Prepare, nothing happens.

voice2001, 2018-03-29:
When I press "Prepare", nothing happens.
Or it can wait, for example, 10 seconds and start slicing. And can wait 20 seconds. Or do not start at all, and you need to restart Cura.
It happens that I press Prepare, nothing happens. I press again "Prepare" and at this second it starts slicing and then cancels it.
The rest of the interface works fine, does not hang. Those I can click Prepare (and nothing happens), then immediately change some value in the parameters or rotate the model in the main window.
Why is this happening?

diegopradogesto, 2018-03-30:
There are some timers that may affect, but in my case the feeling is that it takes maximum 1 sec to start slicing. Please try with the 3.3BETA version that is now available and see if there is any difference.

voice2001, 2018-03-30:
hmmmm.... All versions are shattered...
3.1.0, 3.2.1, 3.3.0-beta
Previously, this was not the case on older versions.

diegopradogesto, 2018-03-30:
What do you mean? After 3.0 comes 3.1 and after that 3.2 and then 3.3 (that will be released in approximately one month). Previous to a release it comes the beta, and this is the status now.

voice2001, 2018-03-30:
I mean that the problem described above only started on version 3.2.1.
But now I started the old version 3.1.0 and the new 3.3.0-beta - and they all showed this problem.
Probably something went wrong in the general settings.

voice2001, 2018-04-01:
I deleted the old settings (renamed the folders with the settings: /home/<user>/.config/cura, /home/<user>/.local/share/cura). Then I set all the settings again. Now the Cura starts the slice every time when I click the Prepare button, but waits about 5 seconds before the start.

closed 2018-04-01 by voice2001

Ghostkeeper, 2018-04-26:
That is strange to me because we never communicate configuration from newer versions back to older versions; only the other way around. Older versions wouldn't understand the newer configuration.

I've seen this bug sometimes too, but it's irregular and I don't know how to properly reproduce it yet.

dsimonet, 2018-09-02:
Same here. By default, any application can't access to internet on my computer. Can't make any slicing if firewall's blocked but it work like a charm went authorized ! I was about to think slice was performed on external server ^^