Ultimaker/Cura #3598: Improvement Request: tweaks/enhancements to Layer View - especially toggles

joyfromprocess, 2018-03-29:
I have a number of enhancements that I'd love to see for **Layer View**. I have no idea how realistic/difficult these changes would be, but wanted to open a discussion. This is my first time submitting an enhancement request, so please let me know how I can improve future submissions.

1. **Allow the user to separately toggle the visibility of walls vs skin**. They currently both fall under "Show Shell" but there are times when I'd like to look at one without seeing the other. They are already colored separately, just not controlled separately. 
2. Make clear in the **color legend** that **retracted travel and unretracted travel are different shades of blue**. I figured this distinction out by accident, but it's a great piece of information to have. I'm guessing that many users may not realize (like me for months).
3. **Allow the user to show/hide z-seam points**. I know I can turn on coasting and that will show me via small gaps, but I don't actually use coasting. This means that I sometimes need to enable coasting to visualize my z-seam but then remember to turn it back off before the final slice.
4. **Scale the Feedrate color scheme** based on **visible** toolpaths. If I'm hiding travels, I don't want travel speed to blow out my scale. 
5. **Allow the user to toggle visibility of the models** (i.e. the light gray ghost that currently remains there regardless of other settings). Especially when I'm using a bunch of modifier meshes, it would be nice to hide them sometimes.

joyfromprocess, 2018-03-30:
I just wanted to add another use case where it would be nice to toggle model visibility (item 5 above).

This example model has a back wall that is too thin to slice, so the "Remove Empty First Layers" feature is (appropriately) dropping the print. The model itself is sitting higher on the bed though, so the gray ghost still matches that. None of this is unexpected or a "bug" - it just makes Layer View more visually confusing if model 'ghosts' can't be toggled off.

[image]

fieldOfView, 2018-03-30:
Regarding your latest comment: There's a setting named "Remove Empty First Layers" in Mesh Fixes that is turned on by default.

joyfromprocess, 2018-03-30:
Thanks - I was expecting that behavior per my comment:

> ...the "Remove Empty First Layers" feature is (appropriately) dropping the print...

I was just using it as an illustration of when the model and slice may be out of alignment.

fieldOfView, 2018-03-30:
'doh! I'm a just-looking-at-the-pictures kind of guy. So, in the second picture, what would you expect? I think it is actually a fairly good indicator that there are unexpected "things" going on in the layers.

joyfromprocess, 2018-03-30:
In the second picture I don't expect it to behave any differently. I agree that it should show it that way by default.

I would just like a toggle that can hide visibility of the model ghost. This was merely a quick-and-dirty example of when it could be convenient to reduce the clutter. 

A more realistic example is when I load a number of modifier meshes onto the same model. It would be nice to have the option to hide the various gray blocks I have jutting in and out of my model/workspace.

Ghostkeeper, 2018-04-26:
1. I concur. We have to be a bit careful though with space. I don't know how much more we can fit in there for users with a limited resolution laptop.
2. Here we're also limited by space. Maybe a good solution would be to make the square for travel moves have two colours? Like separated by their diagonal. It won't tell which is which but at least it shows that there are two colours. We could also do something with a tooltip. @LukeChen-Ultimaker what do you think we can do to improve the legend of Layer View w.r.t. travel moves vs. retracted moves?
3. This is kind of a hassle to indicate with our current architecture. It can be done and it's not hard, but we'd have to implement a new data type to be sent over Protobuf from the back-end to the front-end. I don't think you'll see it coming any time soon.
4. Yeah nice idea!
5. I don't think this is necessary. We render the ghosts such that they are always behind layer view, so they shouldn't be in the way and their colour is distinct enough to not distract (unless you have a custom theme perhaps). Hiding the shadow of objects is a very obscure use case.

I'll create a ticket for 4 now. Once we have a better idea of what to do for 1 and 2 we can plan that too.

joyfromprocess, 2018-05-02:
Thanks, @Ghostkeeper - I appreciate your time and feedback!

Re `#1` 
Not sure if this idea overcomplicates rather than simplifies, but just spitballing:
Maybe there could be a "Simple Layer View" and "Advanced Layer View" as separate drop-down views? 
People who are device/resolution limited (or simply don't care about granularity) could use the Simple Layer View, while detail-hungry people like me could use Advanced. 
Simple Layer View could then theoretically have an even smaller legend than the current Layer View (though admittedly I'm not sure what I would suggest hiding)

Another thought (which I assume would require a more development work... but I honestly have no idea) would be to allow legend items to be selectively hidden like the slicer settings are.

Re `#2` 
If either of my spitball ideas above are practical, they would address this item too.
If my spitball ideas are no good, I like the idea of a diagonal slice paired with a tooltip that explains the difference to the user.

ianpaschal, 2018-07-02:
For reference: JIRA CURA-5298

Ghostkeeper, 2018-07-04:
The ticket for idea 4 was rejected by our project manager. We won't be able to spend time on that one any time soon.

lscarneiro, 2020-05-26:
> The ticket for idea 4 was rejected by our project manager. We won't be able to spend time on that one any time soon.

I came to this feature request for this exact idea (number 4), we're almost two years later, any chance this could be considered now?

My travel speed is 3x the print speed, so I mostly see blue or blue-greenish on my prints.

Would love to be able to have this range adapted to what is selected or simply ignore the travel speed totally

Ghostkeeper, 2020-09-11:
It wouldn't be reconsidered unless there is a change in the supply or a change in the demand of this feature. We're not currently seeing any reason why this is more important now than it was two years ago (no change in demand). And there will hardly ever be a change in supply since we're never really running out of things to implement and fix.