Ultimaker/Cura #3602: Support blocker initial cube size influenced by model scaling

smartavionics, 2018-03-30:
**Application Version**

master branch

**Platform**

Linux

**Qt**

5.8.0

**PyQt**

5.8.2

**Display Driver**

Something!

**Printer**

Kossel Mini

**Steps to Reproduce**

Load small model STL, select mesh and scale up (in this example by 2000% (aka x20)), select support blocker tool and click on model to create support blocker cube

**Actual Results**

Cube is 20 times bigger than it would be if the model had not been scaled.

**Expected results**

Cube should be its normal size.

**Additional Information**

I appreciate that the huge cube can be selected and scaled but if you have to do this for every cube it's rapidly going to get rather boring. Mind you, being able to select the initial cube size would be very nice. How about having a slider and entry field somewhere that can let you set the initial cube size?

[image]

fieldOfView, 2018-03-30:
Already fixed in https://github.com/Ultimaker/Cura/pull/3545

smartavionics, 2018-03-30:
Hi Aldo, that's great. Closing...

closed 2018-03-30 by smartavionics

fieldOfView, 2018-03-30:
Regarding your suggestions in the additional info: My original plan was to be able to click and drag on an area if the model. Dragging would scale the support blocker. The same would work for already placed supports; drag them to resize. Dragging them with the shift or control key could move them around. The tool could remember the scale of the last placed support blocker.

smartavionics, 2018-03-30:
Being able to resize and move already placed blocks would be very nice. I do think that being able to specify the initial size of the blocks using either a slider, entry field or remembering the last size would be very useful because it's a lot easier to place blocks that are already of the correct size (or close to correct) than it is to place blocks that will subsequently need resizing.