Ultimaker/Cura #3603: Materials no longer filtered by compatibility

fieldOfView, 2018-03-30:
**Application Version**
3.3, master post machine manager refactor

**Platform**
any

**Printer**
Eg Cartesio

**Steps to Reproduce**
Add a Cartesio printer
Check listed materials in material menu

**Actual Results**
In 3.3 all known 1.75mm materials from all manufacturers are listed, including materials that will not print well on the selected printer.

**Expected results**
In 3.2 and before, only a set of generic materials and two materials from DSM are shown.

**Additional Information**
Note: This is a problem for me on another printer (Blackbelt), but the Cartesio case is easier to reproduce.
/cc @maukcc

alekseisasin, 2018-04-04:
At this moment in Cura two type of material sorting. For UM machines and Custom FDM printers which shows all generic materials. I agree we need to show sorted list of materials if they are predefined.
Devs: CURA-5196

maukcc, 2018-04-04:
This is also discussed at #3461. But I am guessing that the materials showing up, is a symptom. 
The underlying problem is that we do not have clear guidelines on the relations between files as described in #3461.

fieldOfView, 2018-04-13:
Does this stand any chance of getting attention before the 3.3 release?

ianpaschal, 2018-04-24:
@fieldOfView No not really. We'll be making lots of big changes to materials in the very near future though.

fieldOfView, 2018-04-24:
Ok, but do we agree this is an unintended change? I really don't know anymore, with the exclude_materials metadata that was recently added to UM definitions (CURA-5149 ?)

ianpaschal, 2018-04-24:
Yes. I'm pretty sure it was unintentional.

And we intend to be adding (fixing?) that along with new features like "Favorites" (let's be honest; most people can count the materials they use day-to-day on 1 hand) in the overhaul.

Right now we're not working on it, but when we do this ticket will be marked as in progress.

ianpaschal, 2018-07-02:
On 17 May 2018 we marked this "Won't do" internally since it will be changed with some upcoming reworks to the whole materials interface.

closed 2018-07-02 by ianpaschal