Ultimaker/Cura #3604: Feature Request: Build Plates per machine

cbusillo, 2018-03-30:
Hi, I have a suggestions that would help out my work flow.  Can you add the ability to have the build plate feature per machine?  That way, when I select a new machine, I am able to get the last build plate pulled up.  An easier feature might be to allow the build plates to be renamed.  That way, I would at least know which build plate was on which printer.  THANKS!  BTW Cura 3.3 is awesome.  I'm finally switching from S3D.

cbusillo, 2018-04-04:
Yes! When I use S3D, that’s my work flow. Cura is so much better now, but I miss that feature. Thanks!

ianpaschal, 2018-07-02:
For reference: JIRA CURA-4975

Ghostkeeper, 2018-08-13:
CURA-4975 has been changed into removing the multi-buildplate feature instead of linking it to specific machines.

Ghostkeeper, 2018-08-13:
It takes too much memory to store the models completely, but we could associate file names and transformation matrices with the machine stacks. At least in memory; we don't need to actually store it in the profile I'd say. It's not necessary to re-load the open files when you restart Cura.

jackha, 2018-08-14:
We've discussed this in the team and we're not going to implement this.
The Open recent menu is intended for this.

closed 2018-08-14 by jackha