Ultimaker/Cura #3605: Three perimeter (1.2mm) wall

kondzio29, 2018-03-30:
**Application Version**
3.3.0

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**
It is not related to GUI 

**Printer**
Custom printer setup

**Steps to Reproduce**
I am making a model with 1.2 wall thickness, and in CURA i set wall thickness to 1.2mm, in ver. 2.7 it is working fine, and generates wall made of three perimeters, with no additional infil. But in 3.2.1 and also in 3.3 it generates mess that causes my printer to retract after each dot of wall infill

**Actual Results**
Results produced by version above 2.7.0 including 3.3.0
[image]
those yellow lines causes printer to retract and make a dot of filament, and not onlu the print tooks more time, but I experienced filament jams and failed prints due to that behavior.

**Expected results**
Results that were produced by version 2.7.0 
[image]
Those lines resulted in good prints and none of them failed. 

**Additional Information**
in both versions i used the same settings with 100% infill if that matters. 
I tried resetting settings, reinstalling cura, and wiping the registry without sucess. Only on cura 2.7 i was getting result i was expecting. 

[CFDMP_maly top_curaproject.ZIP](https://github.com/Ultimaker/Cura/files/1864704/CFDMP_maly.top_curaproject.ZIP)

Appesteijn, 2018-04-02:
Could you try to change the 'Maximum resolution' to 0.02 ? And I can't open the zipfile you attached, what is the original extension?

Ghostkeeper, 2018-04-02:
Looking at the magic number, `50 4B 03 04`, it does appear to be a zip archive. 7-zip and Windows get an error when opening this archive though. HxD finds file names in there like `3D/3dmodel.model` which indicates that this is a 3MF file.

I've been thinking that this bug has always been in Cura. It's the natural result of how we create the walls by creating insets from the outer perimeter. You've just found a case where the second inset doesn't fit between the perimeters in the straight bits but does fit in the corners. Maybe some of the settings that relate to the walls were changed in between those two versions (such as Outer Wall Inset).

I haven't been able to reproduce it yet though. I'm now testing with this model: [ThinCylinder.zip](https://github.com/Ultimaker/Cura/files/1869659/ThinCylinder.zip)
I tried saving as CuraProject but that gave an error so I'll just say that I used an outer wall line width of 0.345 and inner wall line width of 0.359 to produce a similar but different effect that seems to be the result of rounding errors. With an inner wall line width of 0.358 or 0.360 the result is as expected. So it seems to be a property of how the model was triangulated. My triangulation was done by OpenSCAD, which doesn't taper the corners in any way.

kondzio29, 2018-04-03:
@Ghostkeeper but how it is that Cura 2.7 can generate such walls using the same model as 3.3? It is exact same STL. I have been using Cura like that for printing walls for a long time without problems. It reduces print time greatly. When cura wants to make that weird infill inside walls it causes a ton of retraction between each dot of extruded filament and finally nozzle blockage.

BagelOrb, 2018-04-03:
Yellow is infill. It might be a problem of the infill overlap setting.

Or it might be perimeter gaps. Does the problem reproduce if Fill gaps between walls is enabled?

Note: your model cannot have a wall thickness of 3 walls, because the model itself is as thick as 3 walls, meaning that it can only fit 1.5 wall on each side. That's why it's having more trouble with this model than if it was 1.6mm thick, which supports 2 walls on each side.

BagelOrb, 2018-04-03:
* if Fill Gaps is _disabled_

Ghostkeeper, 2018-04-03:
> @Ghostkeeper but how it is that Cura 2.7 can generate such walls using the same model as 3.3? It is exact same STL. I have been using Cura like that for printing walls for a long time without [...]

Because you are using different settings in 2.7. You can see in your screenshot that the line width is 0.4mm in Cura 3.2 and 0.38mm in Cura 2.7.

gumert, 2018-04-04:
I reproduced similar behavior in Cura 3.2.1 and 2.7. I tried with a smaller diameter cylinder initially and didn't have any luck. At some point, I thought about trying a larger diameter cylinder and got a similar looking, but slightly different, behavior.

[image]

Interesting, it looks like the behavior is a function of the wall thickness. The top of the wall is 1.2 mm wide, and then tapers after a few layers. Every layer of the taper, aside from one, looks normal. However, one of the layers in the tapered section contains a similar looking anomaly. 

[image]

[STL can be found here](https://www.thingiverse.com/thing:2849537)

Xtrudor, 2018-04-14:
Same here. 1.2 mm wall. Wall thickness 0.8 mm. Line width 0.4 mm. No middle line!
[image]
[image]

lolorc, 2018-04-18:
issue confirmed here as well

Xtrudor, 2018-04-23:
If that helps... Flat rail with 4 mm walls... no walls. Print Thin Walls = ugly wall. Negative Outer Wall Inset looks ok...
[image]
[image]
[image]

lolorc, 2018-05-03:
thought the small negative outer wall inset was not needed anymore. IIRC it was useful to print hollowed objects with wall width==line width

fatalelieberi, 2018-05-19:
I have this problem too when printing my model. In case some data are needed, I'm sharing with you guys the STL, Project, and Setting files.

[MyPrinter.curaproject.zip](https://github.com/Ultimaker/Cura/files/2019059/MyPrinter.curaproject.zip)
[ArcReactorMkI v3_P17009-0008-Arc Base_2.zip](https://github.com/Ultimaker/Cura/files/2019060/ArcReactorMkI.v3_P17009-0008-Arc.Base_2.zip)

[image]

fatalelieberi, 2018-06-27:
Awesome! I tried Cura 3.4 and it worked perfectly! No more rough thin wall surface anymore.

[image]

closed 2018-08-08 by Ghostkeeper

hwgfp, 2019-02-16:
Hi, everyone.
I think I have the same issue with Version 3.6 (i am just starting with cura and 3d printing).  I made a simple cylinder and set the infill to 100 %. Unfortunately, the infill is no continuous circle. 

[image]

.stl file:
[becher.zip](https://github.com/Ultimaker/Cura/files/2871895/becher.zip)

Ghostkeeper, 2019-02-18:
> I think I have the same issue with Version 3.6.0 (i am just starting with cura and 3d printing). I made a simple cylinder and set the infill to 100 %. Unfortunately, the infill is no continuous circle.

Your model doesn't have equal thickness everywhere. You can see it in the screenshot you made. It looks like the inside perimeter has a different resolution than the outside one, causing the vertices to misalign. At the top of the cylinder in your screenshot, you can see that the inside corners of the inner perimeter are about halfway along an edge in the outside perimeter. That makes the perimeter slightly thinner in that part of your cylinder and prevents Cura from fitting another wall in there.

hwgfp, 2019-02-21:
Thank you. I think I exported the mesh not precise enough from FreeCaD. Therefore was my object more angular than round.

rolpal, 2019-03-27:
Thanks to the comment of @Ghostkeeper above pointnig at the accuracy of the perimeters I solved a similar problem.
I want to print a square with rounder corners and three lines of thickness as a part of a design.
I made the wall 1,2mm (0,4mm nozzle ) and wall line count set to 3 (for the sake of the rest of the print). This gave me continuous lines in outer and inner wall but the inside was sliced as diagonal infills.
After changing one of the wall line width to 0,395 the wall became three continuous lines.
Any experimentation with infill pattern, %, wall line counts etc. had no effect.

Conclusion(?): Cura will only use continuous line if it will fitt IN the gap and 0,4 does not fit IN 0,4.

I hope this will add some insight to the problem of not getting continuous lines in walls.

Ghostkeeper, 2019-03-29:
> This gave me continuous lines in outer and inner wall but the inside was sliced as diagonal infills.

You might be encountering this issue: https://github.com/Ultimaker/Cura/issues/5110

> BUT... it gave me another problem with uneven (over) extrusion in the outer wall next to the Z-seam.
Tried different settings but cant fix it but that is not for this thread...

Maybe try adjusting the Coasting settings.