Ultimaker/Cura #3606: [3.2.1] Freezes on model movement

yyh1002, 2018-03-31:
**Application Version**
3.2.1

**Platform**
Windows 10

**Display Driver**
Nvidia 391.35

**Printer**
UM2+

**Steps to Reproduce**
Load a model, and move or rotate the model

**Actual Results**
The software freezes during movement or model selection. It doesn't happen every time, but very frequently. If it didn't freeze during movement, slicing would always succeed. 

**Additional Information**
I didn't had the issue with 3.1 or previous versions at all. 
I'm using Mark2 mods for Cura. 
Recent log file after freezing is attached
[cura.log](https://github.com/Ultimaker/Cura/files/1865179/cura.log)

Thanks

Appesteijn, 2018-03-31:
Could you try 3.3-beta? We did some improvements with regards to the rendering.

yyh1002, 2018-04-01:
I tried 3.3b but it keeps crashing at startup. There seem to be some problem with carrying over the old machine settings. The 'backup and reset configuration' button didn't help. I deleted the user data of 3.3 in Roaming folder but still couldn't get a clean startup without any previous influence.

alekseisasin, 2018-04-04:
@yyh1002 , If you removed all user data and after starting Cura it keeps crashing then please can you upload the log file again.

yyh1002, 2018-04-05:
This time I moved 3.1 and 3.2 data folders elsewhere and were able to successfully launch 3.3b. 
It seems 3.3b is conflicting with the Mark2 mods, but I attached the crash log anyways. I'll just use 3.3b without the mods until the mods get updated.
Thanks for the followups.
[cura 3.3 crash.log](https://github.com/Ultimaker/Cura/files/1878421/cura.3.3.crash.log)

alekseisasin, 2018-04-05:
I see the error happened because of this file: Mark2_for_Ultimaker2_0.8.inst.cfg.
Might be the Version Upgrade has issue if you did not change the settings manually. 
Also If you used before beta versions of Cura they could make files not portable to a new Cura version.

Ghostkeeper, 2018-04-30:
That Mark2 mod seems to contain an HTML page in one of the configuration files. It's not just 3.3 beta though: It's also what caused the freeze in 3.2.

You could try 3.3 stable since we made some stability improvements there for when the user's files are corrupt. But other than that, I don't think it's in Cura's scope to fix it for you or recuperate from this. Stability improvements are an ongoing change.

closed 2018-04-30 by Ghostkeeper