Ultimaker/Cura #3607: Machine Settings start/end G-Code resets to default script when you click out of the box

SwiftNick, 2018-03-31:
**Application Version** 3.3.0-BETA
**Platform** Windows 10 Pro x64
**Qt**     as shipped with beta
**PyQt** as shipped with beta
**Display Driver** Nvidia GeForce 388.73
**Printer**  Vertex K8400 Dual

**Steps to Reproduce**
Select printer>manage Printers>machine settings
Click Printer tab
Click in Start G-Code (or end) box and make some changes
Click between start and end G-Code edit boxes a couple of times

**Actual Results**
Changes made to scripts are lost and scripts revert to default

**Expected results**
Changes should persist

**Additional Information**

cnering, 2018-03-31:
I'm experiencing this too.  As a workaround, I found that if you add a new-line via the ENTER key after you paste you start/end GCODE it keeps it.  I have to do each section by itself though, so my workflow is:

1. Paste whatever I want into the START GCODE section
2. Put a newline at the end (regardless if there is already a newline, you need to put another one)
3. Close the dialog box and quit Cura
4. Reopen Cura and perform the same steps on the STOP GCODE section
5. Finally, close and reopen Cura

That seems to work for me.

nyaray, 2018-04-15:
Just had this happen, with the difference that I tabbed out and back to look up some commands in my browser. Maybe it's a (un)focus event handling issue?

Unfortunately, I hadn't backed up my gcode, so I'll have to try to remember what I had there and put it in a gist or something :(

Ghostkeeper, 2018-05-04:
I can't reproduce this though. Could someone retry on the stable 3.3? We made some fixes in the beginning of April that could have fixed this too.