Ultimaker/Cura #3608: PauseAtHeight plugin:

JPFrancoia, 2018-03-31:
**Application Version**
3.2 AppImage

**Platform**
Archlinux

**Qt**
Irrelevant

**PyQt**
irrelevant

**Printer**
UM2+

**Steps to Reproduce**
Add a pause at a certain layer with the PauseAtHeight plugin present in master

**Actual Results**
The print seems to pause well, but crushes the part being printed ! I think the gcode tries to go to a low position in Z.

**Expected results**
Normal pause resuming

**Additional Information**

Since recently, I was using my own pause at height plugin. But for reasons, I had to switch to the AppImage of Cura, version 3.2. The pause at height plugin embedded in this version is buggy (I think my PR made on the old PostProcessingPlugin repository wasn't merged on time to be integrated to this version). So I decided to take the one from Cura master, which allows to pause at a certain layer, instead of a certain height, which is handy. I found at least two bugs with the plugin on master when pausing at a certain height, but I'll start with the most important one:

I generated a gcode file with a pause at layer 36: 30 mm retraction at the pause, 100 mm extrusion when resuming pause, and redo 1 layer when resuming. The pause should occur at around 10.9mm. Here is the portion of gcode for the pause:

```
;TIME_ELAPSED:5114.883345
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 37
M83
G1 F1500 E-30
G1 F300 Z1
G1 F9000 X190 Y190
G1 F300 Z15
M84 E0
M104 S0; standby temperature
M0;Do the actual pause
M109 S260; resume temperature
G1 F1500 E30
G1 F199.998 E100
G1 F1500 E-30
G1 F300 Z1
G1 F9000 X0.0 Y0.0
G1 F1500 E30
G1 F9000
M82
G92 E2492.85247
;LAYER:36
G0 X97.006 Y100.743 Z12.28
M204 S1000
M205 X10 Y10
;TYPE:WALL-INNER
G1 F600 X98.221 Y99.822 E2492.85247
```

There are two ```G1 F300 Z1``` instructions that bother me. One before the actual pause, and one after. I couldn't see my print pausing (I wasn't in front of the printer when it occured), but it seems it paused fine. I was there when I resumed it: the plate started going up and kept going, and the nozzle burnt through my print. If I'm not mistaken, these two commands put the plate at Z=1. There is nothing to switch the Z axis to relative mode, so it's in absolute mode. And my print is supposed to stop at 10.9 mm.

This is the portion of gcode generated by my "old" plugin, just before the old PostProcessingPlugin repo was merged to Cura's repo. This plugin was tested thoroughly in real life and I know it works:

```
;TIME_ELAPSED:5113.780317
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 12.580000 
;current height: 11.050000 
M83
G1 E-30.000000 F1500.000000
G1 Z13.580000 F300
G1 X190.000000 Y190.000000 F9000
G1 Z15 F300
M84 E0
M104 S0; standby temperature
M0 ;Do the actual pause
M109 S260; resume temperature
G1 E30.000000 F1500.000000
G1 E100.000000 F199.998000
G1 E-30.000000 F1500.000000
G1 Z13.580000 F300
G1 X100.888000 Y119.220000 F9000
G1 E30.000000 F1500.000000
G1 F9000
M82
G92 E2492.895990
;LAYER:36
G0 X102.368 Y97.909 Z12.28
M204 S1000
M205 X10 Y10
;TYPE:WALL-INNER
G1 F600 X103.857 Y97.583 E2492.89599
```

Comparing the two codes, it seems the two ```G1 F300 Z1``` should actually be "current_z" + 1.

This is the code generated by the plugin on master, when setting the pause with height in mm, not layers:

```
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 12.58
;current height: 11.05
M83
G1 F1500 E-30
G1 F300 Z13.58
G1 F9000 X190 Y190
G1 F300 Z15
M84 E0
M104 S0; standby temperature
M0;Do the actual pause
M109 S260; resume temperature
G1 F1500 E30
G1 F199.998 E100
G1 F1500 E-30
G1 F300 Z13.58
G1 F9000 X97.006 Y100.743
G1 F1500 E30
G1 F9000
M82
G92 E2492.89599
;LAYER:36
G0 X102.368 Y97.909 Z12.28
```

So basically, setting a pause based on layer number doesn't work with plugin on master. Setting the pause with layer height works.

I think somewhere in the loop of this file: https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py#L173

```current_z``` is not set properly. Also X and Y at the pause are not set properly either, since the pause resumes with ```G1 F9000 X0.0 Y0.0```. And this only happens if the pause is set with layer number.

Any idea on how to solve the issue ? I might look at it tomorrow but not sure I'll have time.

JPFrancoia, 2018-04-10:
I just made a PR that fixes the issue. Basically by selecting the layer number, ```current_z``` was never calculated. Changing a ```break``` to ```continue``` fixes the issue.

However, raft layers were not accounted for when selecting pause by layer number. Now the plugin does.

Also, now the (X, Y) position of the nozzle at the end of the pause is the (X, Y) position of the nozzle for the layer immediately after the pause. Much better since X, Y of this layer often is not identical to X, Y at the end of the pause.

ChrisTerBeke, 2018-05-24:
Fixed by https://github.com/Ultimaker/Cura/pull/3648.

closed 2018-05-24 by ChrisTerBeke