Ultimaker/Cura #3609: [3.2.1] File saving: filenames cut by first dot

acorderob, 2018-04-01:
**Application Version**
3.2.1

**Platform**
Windows

**Steps to Reproduce**
- Load STL with dots in its filename (f.e. Waving_Groot_15.5cm.stl) (also happens loading projects)
- Project name is cut by the first dot (Waving_Groot_15)
- Fix it manually (Waving_Groot_15.5cm)
- Generate gcode.
- gcode filename gets cut by first dot (Waving_Groot_15.gcode)

**Actual Results**
Filename/project name gets cut by first dot.

**Expected results**
Filename/project name should be kept intact.

alekseisasin, 2018-04-04:
@acorderob , Did you write correct Cura version?

acorderob, 2018-04-04:
Yes.

alekseisasin, 2018-04-05:
The I am a bit confused. I tested with 3.2.1 and I can load the model with dot in file name. It loads and shows proper name, afters saving same still see the name with dot. But in 3.3 Beta it works as you described.

acorderob, 2018-04-06:
I just tested it again with the same files. Maybe i did something wrong because now the cut only happens when opening a project (loading "Waving_Groot_15.5cm.curaproject.3mf" results in project name "Waving_Groot_15" which will then be used as filename when saving).

Ghostkeeper, 2018-04-30:
This is probably very dependent on the operating system, so a developer that uses Windows should be testing this.

alekseisasin, 2018-06-28:
This issue was fixed. Please check Cura 3.4 version.

closed 2018-06-28 by alekseisasin

JesConsuegra, 2019-04-13:
It happens again in Cura 4.0 Win64