Ultimaker/Cura #3610: Cura3.2.1 material parameters for UM2+

2ndClassUser, 2018-04-01:
**Application Version**
3.2.1

**Platform**
win 10

**Qt**

**PyQt**

**Display Driver**
Geforce gtx

**Printer**
Ultimaker 2+

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

alekseisasin, 2018-04-04:
Unfortunately it is correct behavior. Even it was already in Cura 3.1
If will open a Preferencse, in Settings you can find "Initial Printing Temperature", hover mouse cursor over the exclamation mark. You will see that the settings has dependency on 'Enable Nozzle Temperature Control ' and this setting is available for certain Gcode flavour. It means UM2 uses different flavor and the setting is not available for it.

closed 2018-04-04 by alekseisasin

fieldOfView, 2018-04-04:
@2ndClassUser, if you go to Machine Settings on the Printers pane of the Preferences, you can set the gcode flavor to "Marlin" (instead of "Ultimaker 2"). Among other things, this will give you full control over temperatures.