Ultimaker/Cura #3612: (3.3.0-beta) reboot printer

macaron16, 2018-04-01:
**Application Version**
3.3.0 beta

**Platform**
osX High Sierra

**Printer**
Anet A8 on marlin 1.1.x bugfix

**Steps to Reproduce**
Connect the USB cable.
Turn on the printer
Launch Cura

(work also if you start to launch Cura before turn on the printer).

**Actual Results**
The printer reboot, once ... twice... 

**Expected results**
Not reboot at all...

**Additional Information**
No issue with the Cura 3.2.2 beta, with the same firmware.
I just update cura to 3.3.0-beta.

fieldOfView, 2018-04-01:
Cura has done this since version 2.1. Perhaps not twice, but it has always been a problem that Cura resets connected printers.

fieldOfView, 2018-04-01:
See https://github.com/Ultimaker/Cura/issues/1049

cnering, 2018-04-05:
What main board does your printer use?  If it's an Arduino, this is the microcontroller's default behavior to reboot itself when serial connection is established.  It's just how the chip is designed, I don't think Cura can connect without activating the reboot.  There are some guides on how to disable the functionality on the Arduino side, but they all seem to require some hardware modifications.

If you don't actually send print jobs to your printer from your PC via USB, you can disable the USB Printing plugin and Cura won't scan for boards on startup, which won't establish a serial connection, which in turn won't reboot your board.

macaron16, 2018-04-05:
Hi,

My electronic board is the original one (Anet V1.5).

cnering, 2018-04-05:
I can't find detailed specs for your specific board, but the pictures I'm seeing are all Atmel ATMEGAs so it would make sense that this is the hardware issue you're seeing.  If you're really interested in fixing it, this looks like your best bet, but be advised that it looks like you'll need to do some exploration and experimenting to get it working:

https://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection

ChrisTerBeke, 2018-04-05:
Anet has the most horrible piece of electronics in existence. Mine reboots all the time, regardless of which software I use to control it.

ChrisTerBeke, 2018-04-05:
As for rebooting when connecting USB, that's perfectly normal, as a connection has to be set up between Cura and the USB (serial) device. Marlin Firmware can't do that unless being restarted to 'switch hosts'. To fix that, Cura shouldn't connect to the USB device, but als @fieldOfView incidates, there's already a ticket for that.