Ultimaker/Cura #3615: [REQUEST] Allow combing over thin sections.

Douggie-J, 2018-04-03:
A lot of my prints are mechanical parts that have (or are almost entirely made of) thin walls that need to be fairly clean and free of blobbing. Combing would solve the issue, but combing does not work if the layer doesn't have at least 1.5-ish inner walls to travel over. An option to allow combing over outer walls would be extremely useful for parts like this, as the only other "good" option is to use a very fine nozzle; an issue in itself.

Douggie-J, 2018-04-03:
UPDATE: Actually, after some more fiddling, apparently this is already supported, but the behavior is a bit...strange.

Image one shows what I was initially working with. Mess of travels outside the part to fill in the odd cavities formed by the threads of the cap. Image two shows the exact same cap with the wall thickness set to the width of the nozzle, such that there will be no inner wall at all. It now slices exactly as expected.

Solves my problem, at least for this particular component...
[image]
[image]

Ghostkeeper, 2018-04-26:
Yeah we've also seen this bug and we're investigating it now.

Ghostkeeper, 2018-04-26:
In fact, this is a duplicate of https://github.com/Ultimaker/Cura/issues/3458.

closed 2018-04-26 by Ghostkeeper