Ultimaker/Cura #3616: Extruder not retracting during waiting time when "Lift head" is ON.

D4rw1n0zor, 2018-04-03:
**Application Version**
3.2.1

**Platform**
Firmware: Marlin 1.1.7
GCode sent to printer through OctoPrint 1.3.6 running on OctoPi 0.14.0. 

**Printer**
Tevo Tornado

**Steps to Reproduce**
Enable "Lift Head" option in Cooling parameters.
Set Minimum Layer Time to 60 for instance.
Set Minimum Speed to 5mm/sec (or lower).

Print a 2x2mm surface tower (any shape scaled without keeping proportions in cura will do the job).

**Actual Results**
Nozzle moves out from the print (as intended), but does not retract. Filament is still going out and causing first spot of upper layer to get that filament sticking on the edge.

**Expected results**
Provide an additional option (or default) to retract filament from X mm and therefore prevent that problem.

Ghostkeeper, 2018-04-26:
I can't reproduce this. It does retract for me. Here's a snippet of g-code that I get when following these steps:
* Load a 1cm cube.
* Enable Lift Head.
* Set Minimum Layer Time to 100s.

The g-code:

    G1 F300 X113.253 Y111.273 E36.17033
    M204 S5000
    M205 X30 Y30
    G0 F15000 X120.55 Y111.55
    ;Small layer, adding delay
    G1 F1500 E29.67033 ; <-- Here is a retraction!
    G0 F15000 X120.55 Y111.55 Z4.07
    G0 X106.407 Y97.407
    G4 P44798
    ;TIME_ELAPSED:514.052410
    ;LAYER:5

alekseisasin, 2018-06-28:
@D4rw1n0zor , The issue is open for long time and we cannot reproduce it. Could you please check again otherwise we will have to close the issue.

closed 2018-08-09 by Ghostkeeper