Ultimaker/Cura #3619: [3.x] Feature request: option to not create nor update settings when loading project

acorderob, 2018-04-04:
I would like to have the option to ignore project settings (printer, profile, and/or material) when opening a project. And by that i mean using the selected printer/profile/material of the project but not update the current settings (nor create a new copy which i will have to delete later).

Most of the time i have fiddled with the settings and just want an old project to use the new ones.

alekseisasin, 2018-04-04:
When you load a project it has two options "Open as project" and "Import models". If you select import models then you should only load models.

acorderob, 2018-04-04:
I have set it up to always open as project. And what i mean is after it starts loading the project it asks if you want to replace or create new profiles (for each of the setting sets - printer,profile,material). And i would like to have a third option in those select boxes "ignore settings" so the project does not change the current settings but instead uses them.

alekseisasin, 2018-04-05:
It is current work flow in Curra. Might be you can choose "Always import models"

ianpaschal, 2018-04-06:
So something like a tree graph of what's included in the Cura project, and uncheck boxes for which parts you do or do not want to import?

I kind of like the idea but it will be up to our product owner if this is a feature we want to build.

acorderob, 2018-04-06:
Interesting, but that might be too much :)

I just asked for an extra option in the combos on the load project dialog, so the specific option set (f.e. the material) is used but not the specific settings inside the project. Something like this mock up (the last option in the combobox):

[image]

ianpaschal, 2018-04-06:
Under the hood it would be the same thing and I'd rather, in this case, just let the user either pick what they want to import, or import everything, rather than add a new option knowing that in the future a user might want another slightly different import behavior, and another wants something else, etc.

Ghostkeeper, 2018-05-01:
You meant that it does take the correct quality profile and material profile including all their settings, but doesn't apply user settings?

closed 2023-05-14 by acorderob