Ultimaker/Cura #3621: Cura 3.2.1 hangs during Slicing

MasterTinkerer, 2018-04-05:
Cura 3.2.0, Cura 3.2.1 and Cura B3.3 - MacOSX 10.11.6 ( ElCap ) - Lots of RAM - SSD Boot - NVIDIA Quadro 4000 + NVIDIA GeForce GT120 ( MacPro Tower )

I have been using Cura 3.2.0 successfully for weeks. Today I had to run some updates on MacOSX for the sake of  " security " as was suggested by APPLE Inc. Cura 3.2.0 started acting up. After pressing the " Prepare " it pauses longer than usual and then displays " Slicing ... " and never comes out of that loop. 
I Updated to Cura 3.2.1 after removing every trace of the Cura on my system ( past experience ). Problem persists. Same in Beta 3.3
Activity monitor of MacOSX doesn't show any CPU activity while Cura is stuck in slicing...

Macflame, 2018-04-06:
I have a similar problem in Windows 10 Home. I originally got around the problem by running in Windows 8 mode but now the slicer is hanging again. A reinstall 'cures' it until I reboot.

MasterTinkerer, 2018-04-06:
Good to know @ Macflame. One would wonder if there are more faulty Cura systems out there and people don't bother to report.
I removed all traces of Cura from my MacPro and went back to Cura 2.7 ( 32bit ) with the hope of getting some work done. same problem.
The Slic3r  and it's Prusa edition flavour are blunt tools in comparison to Cura but they remain stable and reliable throughout the generations. I'm not sure why Cura has so many bugs and glitches after all these years.

MasterTinkerer, 2018-04-07:
Good News!
Cura 3.2.1 and it's engine works on a freshly installed MacOSX.
I cloned a clean install of MacOSX 10.11.6 ( El Capitan ) then proceeded to installing Cura 3.2.1 and it works as expected even noticeably faster than Cura 3.2.
So, perhaps the cause of hanging slicer engine on my system has been either corrupted permissions on the boot drive or more likely other installed softwares effect the functionality of Cura.
I will be reinstalling my work softwares one by one while testing the Cura. As soon as I find the culprit, I'll report back.

Macflame, 2018-04-07:
@Master Tinkerer I am glad you have now got it running. I have tried changing permissions on the Cura engine but that has not helped. It does the same thing in the Wanhao edition v1.1 (same engine?). I have given up on it for now. I may just install an earlier version later.

MasterTinkerer, 2018-04-10:
Well, the case is closed but Cura 3.2.1 stopped slicing after a Mac shutdown. After reboot I went through all tricks in the book to no avail. 
Today I managed to get some work done with Slic3r Prusa Edition. If this issue persists on stable version of Cura 3.3 I will spend the cash on Simplify3D and be done with it.

Macflame, 2018-05-03:
I shall give that a try Ghostkeeper. Many thanks.

jackha, 2018-07-02:
Any updates on this?

nallath, 2019-04-04:
Closing this due to inactivity, I'm assuming this has been fixed.

closed 2019-04-04 by nallath

Ghostkeeper, 2019-04-05:
The speed of a slice starting has been improved somewhat in 4.0. It's not quite there yet though.