Ultimaker/Cura #3622: [3.x] Request: set speed for small holes

userosos, 2018-04-05:
I want to regulate speed for small holes. Set diameter and speed in the hole and something else.

Ghostkeeper, 2018-04-30:
Why do you want to regulate the speed for small holes? Is there some print failure you're trying to fix?

userosos, 2018-04-30:
@Ghostkeeper if i have an big model with some small holes the holes can print very faster and badly.

Ghostkeeper, 2018-05-25:
What is it about small holes that causes it to break though? Is it the rapid starting/stopping? The sharp corners? The bad adhesion in the first layer?

My first thought is that you could set the speed for small holes by reducing the acceleration and jerk. You're probably not even reaching the maximum speed in these small holes because of your current acceleration/jerk settings.

userosos, 2018-05-25:
@Ghostkeeper I has bad adhesion in the first layer (if one perimeter).

jackha, 2018-07-02:
No reaction in a long time, closing.

Ghostkeeper, 2018-08-13:
Uh, so, reducing print speed for small features, is that something that we think could improve print reliability?

userosos, 2018-08-13:
@Ghostkeeper i think it will work good for big parts with small holes.

alekseisasin, 2018-08-20:
@userosos , The workaround for your issue will be to apply the setting "Initial Layer Speed". It helps to improve adhesion on the first layer. You can try other settings for the "Initial layer". Please check the settings in "Speed" settings group.
If you will need to change the speed for other layers you can use "Post Processing" plugin and "ChangeAtZ 5.1.1 Experimental" script.

Sorry. For now, we can not implement "regulate speed for small holes".

yodor, 2018-10-24:
I'm also interested to have speed for smaller holes. Atleast just for initial layer.

smartavionics, 2018-10-25:
Hello @yodor, my Cura builds support those features (can set speed for first layer walls and also speed for small features). You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

closed 2023-07-13 by github-actions[bot]