Ultimaker/Cura #3625: [3.3.0-Beta] Non Uniform Scaling % X Z invert

Robyrob, 2018-04-05:
**Application Version**
3.3.0-Beta

**Platform**
Windows 10

**Steps to Reproduce**
If I turn off "Uniform Scaling" and scale X (red) from drawing, X % change value, but also Z% change value. If I move the blue (Z) the same error: X % change.
And some times I move the red (X) and only Z% value change.
Also if I write a number in X % field and convalidate with Enter, the value in Z will change.

AbeFM, 2018-05-13:
I had issues with this. Quitting and restarting Cura (at minimum sometimes) fixes it.

Ghostkeeper, 2018-05-15:
Our project manager removed this from our planning since we don't think it's important enough to spend time on now. I suppose it's best to depend on the mm measurement rather than the % scale for now.

BillyGeat, 2018-07-02:
This (really annoying) bug still exist in v3.4... :-(
And to make it more annoying, it happens also if mm measurement is used!

AbeFM, 2018-07-02:
You'll get the software off into left field, it's hard to get back.

I've noticed weirdness with objects that are "grouped", even though there's
nothing else there. Ungrouping them helps but you have to do it yourself,
UM is uninterested in things that go wrong because of it, you just have to
know to ungroup it.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

UltiArjan, 2018-07-02:
reported again on the forum, fyi;

https://community.ultimaker.com/topic/23557-issues-when-scaling-object/?do=findComment&comment=213940

diegopradogesto, 2018-07-18:
This is in our board again.

jackha, 2018-07-18:
Changing separate dimensions does not change numbers on other axis anymore, so original issue fixed. However after rotation the scale numbers may look a bit weird because of the scalar value decomposition.

ianpaschal, 2018-08-01:
We reverted the old fix because it was still not behaving as it should and created more issues. Back to "in progress!"

lekv, 2018-11-15:
I'm seeing a similar issue on 3.6.0 with support blocker scaling. To reproduce, apply rotation to a mesh object (e.g. 90deg), then select it, use the scale tool, uncheck "Uniform Scaling" and pull on each handle. You'll find that two of the axis are swapped, e.g. pulling on the Y handle scales the Z axis. It seems that the rotation applied to the object does not get applied correctly to the blocker.

Is this the same issue?

diegopradogesto, 2018-11-19:
Yes @lekv it looks like it is the same issue. We still keep the ticket in our backlog. Hopefully we will start soon with a small refactor to fix this and improve the performance.

mooklepticon, 2019-03-16:
Still happening in 3.6. I have a bolt standing up and the od is a little large. I want to shrink in X and Y but not Z.

Step 1: start
x 100
y 100
z 100

Step 2: Reduce X to 98, Z also goes to 98
x 98
y 100
z 98*
*not intended

Step 3: Reduce Y to 98. Works fine.
x 98
y 98
z 98

Step 4: Increase Z to 100. Also increases X to 102.04%
x 102.04*
y 98
z 100
*Not intended

I can repeat this, distorting the model further and further. This is more than just X and Z being linked. When changing Z up by 2, X went up by 4.04.

Ghostkeeper, 2019-04-14:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4136. Even though this was earlier, I'm going to close this one because the other one has more information.

closed 2019-04-14 by Ghostkeeper