Ultimaker/Cura #3627: Adding M42 gcode to layer 2 on every slice?

PicoPi, 2018-04-06:
This is not a bug. I just don't know where else to turn to.
I want to add an M42 P6 S255 to the start of layer 2 on every file I slice with cura.
How can I do that?
There is a place for start and end gcodes, but nothing in between.
I know there must be a way, but after a few hours, I give up. 
Thanks for any help.

smartavionics, 2018-04-06:
You can do this with the search and replace post-processing script. I have tested this and it works. There's something broken with entering text in Cura at the moment so it tends to forget what you have just entered into a field.

[image]

fieldOfView, 2018-04-06:
Note that the Search in @smartavionics' example will only detect the first layer change. I would use this instead:

Search:
```;LAYER:```
Replace:
```M42 P6 S255\n;LAYER:```

smartavionics, 2018-04-06:
Hi Aldo, the OP asks for the M42 to be output on the start of layer 2 not on all layers after layer 1 which is why I explicitly match that layer number. I agree that my solution leaves out the layer:1 comment.

fieldOfView, 2018-04-06:
Doh, better brush up on my reading skills...

smartavionics, 2018-04-06:
BTW, I tried using ;LAYER:1$ as the search string but it didn't seem to understand $. Is that right?

fieldOfView, 2018-04-06:
The string that is being matched is the whole layer, consisting of multiple lines. "$" matches the end of the string, not the end of a line.

smartavionics, 2018-04-06:
Ah, thanks, that makes sense.

PicoPi, 2018-04-06:
omg, Thank you. Works like a charm!

closed 2018-04-06 by PicoPi

PicoPi, 2018-04-06:
It works, but as you said, it forgets!
Is there any way to fix the code? 
As it is, it would be no different from opening the gcode in Notepad++ and doing a search and replace manually! Wanted to avoid doing that for each and every file I slice.

PicoPi, 2018-04-06:
Where do I get the 3.3? I found 3.3 Beta, but not 3.3!

smartavionics, 2018-04-06:
3.3 isn't released yet.

PicoPi, 2018-04-06:
me so sad :(

PicoPi, 2018-04-06:
I downloaded the 3.3 Beta for shits and giggles, and also out of desperation. I typed it slow, and now it remembers, and is no longer suffering from amnesia !! Yey
Do you happen to know which python source is responsible for this function?

smartavionics, 2018-04-06:
The amnesia problem is in Uranium, the commit that fixes it is:

https://github.com/Ultimaker/Uranium/commit/7b53e35fcb7f28ca4cda43a79102f5bfee44106d

PicoPi, 2018-04-06:
Dude! u d man.