Ultimaker/Cura #3628: Lock model still moved by other model

BagelOrb, 2018-04-06:
**Application Version**
3.2.1

**Platform**
Linux Mint

**Printer**
UM3, UM2

**Steps to Reproduce**
* load model
* lock model
* load second model
* move model over first model

**Actual Results**
first model is pushed free

**Expected results**
first model doesn't move

**Additional Information**
I have drop on build plate and keep apart enabled

Ghostkeeper, 2018-04-30:
Can't reproduce this bug. Whenever you move the second model over the first model, it only moves the second model outside of the first. The first stays in place (also when not locked).

BagelOrb, 2018-04-30:
[image]

BagelOrb, 2018-04-30:
3.3.0 AppImage:
[image]

BagelOrb, 2018-04-30:
3.3.0 project file:
*remove* extension
[lock_model_bug.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1960376/lock_model_bug.curaproject.3mf.zip)

BagelOrb, 2018-04-30:
The reproduction scenario needs a third model.

diegopradogesto, 2018-05-09:
That is still happening, even with just 2 models in the build plate. CURA-5347

diegopradogesto, 2018-06-12:
This issue is fixed in master now.

closed 2018-06-12 by diegopradogesto