Ultimaker/Cura #3629: Special modes -> surfaces mode - surface acting strange

rooiejoris, 2018-04-06:
**Application Version**
Cura 3.2.1

**Platform**
MacOS 10.11.6

**Printer**
see project file attached

**Steps to Reproduce**
make a mesh
and use the settings as in the project file, or basically use follow mesh surface

**Actual Results**
Sometimes very strange travel moves. [And I have seen and heard sometimes lines are double]
If it is a closed mesh, I do get the desired result, but I would expect with an open mesh that it would travel towards the beginning or just reverse the path. Don't know what would be best, maybe it should even be an option...

**Expected results**
It should just generate a line which is following the mesh surface. It works great in the 'old' cura, the feature was called "Only follow mesh surface"

**Additional Information**
See screenshot with the excessive travel moves [you shuold see none if it would follow the mesh]
I use and abuse this functionality to really control the print moves on some parts. You get complete freedom and control in the print strategy. Also nice to experiment with custom infill paterns. Would be very nice if this could also be implemented for infill meshes

Vermeld erbij dat je met die feature volledige vrijheid hebt over waar materiaal neergelegd w

[image]

[two_project_files.zip](https://github.com/Ultimaker/Cura/files/1884758/two_project_files.zip)

Ghostkeeper, 2018-05-01:
Wow, slicing that thing took like half an hour on my computer.

Ghostkeeper, 2018-05-01:
It looks like the lines are not properly attached to each other in this mode. Normally this is done when creating the polygons, but in surface mode we don't create polygons. It seems to behave as if every sliced triangle is a separate line.

http://dulek.net/work/breaking_lines.webm

jackha, 2018-07-03:
So there is actually a thing which is the same as "Only follow mesh surface": In Surface Mode you choose Surface. But as @Ghostkeeper said, on this model it breaks up in smaller pieces. Although in my slice it jumps only a few times per layer instead of all the time. 

@rooiejoris can you try to print with it and see if that helps?

btw awesome application :-)

rooiejoris, 2018-07-09:
Hi @jackha,

I tried Surface, but it acts different than the 'same' feature as in the old cura's.
That exactly is the problem... : )

cheers / joris

jackha, 2018-07-09:
I'm currently printing one of those, see what comes out.

jackha, 2018-08-01:
Also as part of another issue, we've greatly reduced the "jumping around". Only on a few layers I see a single jump.

For reference: CURA-5317 CURA-5455

closed 2018-08-01 by jackha