Ultimaker/Cura #3630: feature request: initial bottom layer thickness

rooiejoris, 2018-04-06:
**Feature request**

initial bottom layer thickness, make the bottom thickness 0 so you will start with printing the infill at the build plate.

Many times I don't need a solid bottom, sometimes I want to fill the object with plaster afterwards, or I like the view of the infill at the bottom, especially with the Cubic [or other '3D' infill] patterns.
Now I can set the bottom thickness to 0, but if you have some overhang half way, you will get holes.

**Additional Information**
Somebody helped me to get started, but a little bit more attention was needed... : )
See below the changes for the cura engine and the json files [please note at that time the name of the feature was "build_plate_bottom_layer_count" I think  "initial_bottom_layer_count" is better since you [at least I do] search on "initial" in Cura to control all the initial layer settings.
The code below works, but the infill doesn't start the 'corrected layers'.

dmprinter.def.json:

+                                },
+                                "build_plate_bottom_layer_count":
+                                {
+                                    "label": "Bottom Layers on Build Plate",
+                                    "description": "The number of bottom layers directly touching the build plate.",
+                                    "minimum_value": "0",
+                                    "minimum_value_warning": "2",
+                                    "default_value": 6,
+                                    "type": "int",
+                                    "value": "999999 if infill_sparse_density == 100 else math.ceil(round(bottom_thickness / resolveOrValue('layer_height'), 4))",
+                                    "limit_to_extruder": "top_bottom_extruder_nr",
+                                    "settable_per_mesh": true

CuraEngine:

diff --git a/src/skin.cpp b/src/skin.cpp
index 972dd26..5d4fe92 100644
--- a/src/skin.cpp
+++ b/src/skin.cpp
@@ -63,6 +63,7 @@ SkinInfillAreaComputation::SkinInfillAreaComputation(int layer_nr, const SliceDa
 : layer_nr(layer_nr)
 , mesh(mesh)
 , bottom_layer_count(mesh.getSettingAsCount("bottom_layers"))
+, build_plate_bottom_layer_count(mesh.getSettingAsCount("build_plate_bottom_layer_count"))
 , top_layer_count(mesh.getSettingAsCount("top_layers"))
 , wall_line_count(mesh.getSettingAsCount("wall_line_count"))
 , skin_line_width(getSkinLineWidth(storage, mesh, layer_nr))
@@ -241,6 +242,15 @@ void SkinInfillAreaComputation::generateSkinAndInfillAreas(SliceLayerPart& part)
  */
 void SkinInfillAreaComputation::calculateBottomSkin(const SliceLayerPart& part, int min_infill_area, Polygons& downskin)
 {
+    if (static_cast<int>(layer_nr - bottom_layer_count) <= 0)
+    {
+        if (layer_nr >= build_plate_bottom_layer_count)
+        {
+            size_t layer_below = 0;
+            Polygons not_air = getWalls(part, layer_below, bottom_reference_wall_idx).offset(bottom_reference_wall_expansion);
+            downskin = downskin.difference(not_air);
+        }
+    }
     if (static_cast<int>(layer_nr - bottom_layer_count) >= 0 && bottom_layer_count > 0)
     {
         Polygons not_air = getWalls(part, layer_nr - bottom_layer_count, bottom_reference_wall_idx).offset(bottom_reference_wall_expansion);
diff --git a/src/skin.h b/src/skin.h
index a601950..4f65425 100644
--- a/src/skin.h
+++ b/src/skin.h
@@ -142,6 +142,7 @@ protected:
     const int layer_nr; //!< The index of the layer for which to generate the skins and infill.
     SliceMeshStorage& mesh; //!< The storage where the layer outline information (input) is stored and where the skin insets and fill areas (output) are stored.
     const int bottom_layer_count; //!< The number of layers of bottom skin
+    const int build_plate_bottom_layer_count; //!< The number of layers of bottom skin touching the build plate
     const int top_layer_count; //!< The number of layers of top skin
     const int wall_line_count; //!< The number of walls, i.e. the number of the wall from which to offset.
     const int skin_line_width; //!< The line width of the skin.

Ghostkeeper, 2018-05-01:
This was also requested from several people in the community event this winter. It's currently in our top-40. There's a good chance that we'll implement this, but it'll take a while to get there.

ianpaschal, 2018-07-02:
For reference: JIRA CURA-5009

Ghostkeeper, 2020-01-13:
This is implemented for the next Cura version! Finally. Thanks to @BagelOrb for the contribution.

closed 2020-01-13 by Ghostkeeper