Ultimaker/Cura #3632: [UI] Illegible settings in Cura 3.2

ziogianni, 2018-04-07:
Hello Guys, it has been a while since the first time I experienced an issue with the Cura User Interface.

The settings on the right panel turns out to be illegible since the first release of 3.2, issue still reiterated with the 3.2.1 version: Text is kinda broken.

I tried to reinstall the application but nothing changed, so I suppose it should be tied with the OSX Cocoa engine.

Any suggestion?

Thanks

**Application Version**
CURA 3.2.1

**Platform**
OSX 10.13.1

[image]

fieldOfView, 2018-04-07:
This has been fixed in Cura 3.3, which is currently in public beta

ziogianni, 2018-04-07:
Ok thanks for the heads-up.

24c, 2018-04-08:
..but @fieldOfView FYI, there is an issue in the Plug Ins UI now in 3.3, that @Andreaux noticed today. I have it too, using macOS High Sierra 10.4
[image]