Ultimaker/Cura #3634: Loading a freshly generated gcode puts the model on a different location on the bed

MB3Y, 2018-04-08:
**3.2.1**

**Win10**

**Predefined Prusa i3 MK2**

**Steps to Reproduce**
- Slice any stl, slice and save it (it will by default be auto centered on the bed)
- Open file and load the gcode

**Actual Results**
The object is visualised as if the XYZ 000 coordinate is at the center of the bed rather than in the front left corner.
Touching the option "Origin at center" in the Machine Settings tab doesn't make any difference.

This latter makes me believe the  loading and visualisation of gcode is always done at relatively to the center of the bed, regardless of the "Origin at center" in the Machine Settings - while it should take it into account imho.

**Expected results**
When loading a gcode it should be visualised using the current printer settings and take the origin into account.

BagelOrb, 2018-04-19:
Can confirm this happens on UM3s as well. The bug happens for me in about 10% of the times that I load a gcode.

Ghostkeeper, 2018-05-08:
:( Haven't been able to reproduce this so far.

I do reproduce something that might be related, with these reproduce steps:

* Load a model.
* In the machine settings, enable Origin at Centre.
* Slice and save the g-code. You now have g-code that contains negative coordinates.
* In the machine settings, disable Origin at Centre.
* Load the g-code. It now positions at the centre of the build plate instead of in the front left corner. Since it has coordinates around zero, I'd expect the g-code of the object to appear near the front left corner.

BagelOrb, 2018-05-08:
I especially noticed this when the gcode was large - as in prints that take a two days or so.

Ghostkeeper, 2018-05-09:
Found it and fixed it. Turns out that it was reported by one of our QA engineers as well, a month ago.

The issue was that it was trying to detect whether the g-code was sliced for a printer with the g-code origin in the centre. It was detecting that by seeing if you had any move to a negative coordinate. Ultimaker's newest printer, the S5, has their prime coordinates in negative X space so for that printer it occurred every time.

So if you have negative coordinates in your g-code but the printer is not `machine_center_is_zero`, this bug would occur.

I fixed it by placing the g-code in the position that the currently active printer would print it, regardless of which printer was active when it was sliced. Makes more sense to me.

closed 2018-05-09 by Ghostkeeper

BagelOrb, 2018-05-10:
Makes more sense to me too!

Thanks!

commit 6772dc9e (2018-05-09): Show g-code in position it would print

Rather than trying to retrieve from the g-code what the position was that it was originally sliced. It was trying to find whether it was sliced for a printer with center_is_zero by seeing if there are any negative coordinates in the g-code, which was faulty. Now we don't even try to do that any more. We just find where the print would end up if it were printed with the currently selected printer.

Contributes to issue CURA-5068 and fixes #3634.