Ultimaker/Cura #3635: Integrating Ultimaker Cura with Google Home

ziogianni, 2018-04-08:
Hello guys,

ChrisTerBeke, 2018-04-09:
You'd probably need an always-running Cura instance with a plugin to connect to Google.

ianpaschal, 2018-04-09:
It's a really interesting idea but I can guarantee you this isn't anywhere on our radar for a very long time to come. I don't think they're nearly as mainstream as you feel, but you can imagine as a 3D printing company we like to imagine ourselves in the future.

That being said, we review/consider all pull requests so if you're willing to make a proof of concept there's like an infinitely higher chance that this might become a reality.

ianpaschal, 2018-04-09:
@ChrisTerBeke Come to think of it, I'd actually use some outside-of-cura automation to search Thingiverse for X, download the file, unzip, and then open the first .stl using Cura.exe.

ChrisTerBeke, 2018-04-09:
Feel free to hit a certain API with your Thingiverse models and slice requests :P

ziogianni, 2018-04-09:
Let's think about it. Google Home should already have in place the API to implement an automatic system like I envisioned. Moreover there's a well-know third-party app, IFTT that allows the creation of recipes which triggers automatic actions. 

IFTT already has a full compatibility with the google assistant.