Ultimaker/Cura #3636: [3.3-beta] Improvement Request: Moveable and Scalable Support Blocker

fatalelieberi, 2018-04-08:
**Application Version**
Cura 3.3 beta

**Platform**
Windows 10

**Printer**
Tevo Tornado 

**Actual Results**
The support blocker is rather difficult to use. When I want to block some spots from any support, the placement of the block is rather difficult to be predicted. Also, when I put it near the bottom of the model, it causes the model to be lifted and more supports are generated for the lifted part.

**Expected results**
The placement of the blocker should be easy. It can be very helpful if there's a preview where the blocker will be placed. At least, it will be great if the blocker cube can be moved and scaled just like regular model. Also, please don't make the part lifted if the blocker cube is close to the bed.

**Additional Information**
[image]

smartavionics, 2018-04-08:
> At least, it will be great if the blocker cube can be moved and scaled just like regular model. 

You can do that, just select the move or scale tool and then click on a blocker cube and you can move or scale it.

fatalelieberi, 2018-04-09:
Damn, I didn't notice that. I thought I couldn't move it because each time I clicked on the cube, it just disappeared (removed).

fieldOfView, 2018-04-09:
>  Also, when I put it near the bottom of the model, it causes the model to be lifted and more supports are generated for the lifted part.

A general tip: don't hide a bug report in a feature request. I'll see what I can do to fix this before the 3.3 release.

fieldOfView, 2018-04-09:
Regarding the feature request: My original plan was to be able to click and drag on an area if the model. Dragging would scale the support blocker. The same would work for already placed supports; drag them to resize. Dragging them with the shift or control key could move them around. The tool could remember the scale of the last placed support blocker.

fatalelieberi, 2018-04-15:
> A general tip: don't hide a bug report in a feature request. I'll see what I can do to fix this before the 3.3 release.
Sorry about that, I will remind myself make a new thread if I find a new bug.

> My original plan was to be able to click and drag on an area if the model. Dragging would scale the support blocker. The same would work for already placed supports; drag them to resize. Dragging [...]
That's a good idea so user can manipulate the blocker intuitively as long there's a Help Pop-up box. However, in my opinion, dragging to scale seems counter intuitive, people are used to moving things by dragging. So, it might be better to assign dragging as a move command.

smartavionics, 2018-05-18:
Hi, I must have missed the update, where did this get fixed/solved?

ianpaschal, 2018-05-18:
From fieldOfView:

> Update: Fixed in #3545

smartavionics, 2018-05-18:
Ah, OK, I was thinking of some other changes for the support blocker that had been discussed.

ianpaschal, 2018-05-18:
If anyone wants to open a new issue specifically for those things that's fine but the original question was answered and improvements were made so to me that qualifies as closable. And this sort of proves the trick of not knowing if an issue/ticket is "done" if it mutates too much along the way.

fieldOfView, 2018-05-18:
The original request was NOT implemented/fixed. I’m still planning to do so, but would request to keep this issue open untill the original request is implemented (or rejected)

ianpaschal, 2018-05-18:
Okay thanks for clarifying.

ianpaschal, 2018-07-02:
I'm a bit confused but the original request for "Moveable and Scalable Support Blocker" is very much existing (thank you Aldo). Additionally, I can't recreate the bug of the support blocker pushing the model off the build plate (well, it does, but corrects).

closed 2018-07-02 by ianpaschal