Ultimaker/Cura #3637: Add quick Rotate feature

terryspitz, 2018-04-09:
I almost always have to rotate objects by 90, 180 or 270 degrees when importing STL from Fusion360.  It would be really useful to have Quick Rotate buttons on the Rotate toolbar to do these quickly.

ChrisTerBeke, 2018-04-09:
For reference: https://github.com/Ultimaker/Cura/issues/2779. It's hard to implement.

greghodson, 2018-04-10:
In Simplify they have "Place Surface on Bed" I Click Ctrl+L then I click on the surface I want on the bed. I use this probably more than 90% of the time as my rotate method in S3D.

ChrisTerBeke, 2018-04-10:
There's a plugin for that already in the plugin browser.

ianpaschal, 2018-04-10:
I'm going to say this is a won't do. Adds lots of UI clutter to save 500 ms of mouse drag.

closed 2018-04-10 by ianpaschal

greghodson, 2018-04-10:
> There's a plugin for that already in the plugin browser. 
@ChrisTerBeke are you referring to the "Place surface on bed"?- I would be very happy if there is! I had already  looked and couldn't see any description that sounded like that. 

If indeed you were referring to my comment then can you say which plugin it is? 
[image]

Thanks,
Greg

ChrisTerBeke, 2018-04-10:
It's auto orientation yes.

terryspitz, 2018-04-10:
Is there really one that lets you pick a surface to lie flat? That's
different to it picking the orientation itself.

ianpaschal, 2018-04-10:
> My request would be for 3 buttons on the rotate toolbar to rotate
by 90 degrees around the X, Y, Z axes.

That will make the panel far bigger, and will mean lots more mouse clicking to get the result, especially if you want to go 90º the other way.

> If you can put icons for this into the main UI too even better.

Definitely not going to happen. Sorry. Maybe some tweaks to the rotate gizmo to make it less finicky (although I don't really think it is), but we're not going to add these extra buttons.

greghodson, 2018-04-10:
@ChrisTerBeke I am not sure we are talking about the same feature...but I have created a rift here as it is really a different feature from what was requested, so I will rather make any further comments about the Place surface on bed in this  #ticket...  #3644

terryspitz, 2018-04-11:
Thanks for the comments Ian. I wonder if it's worth me looking at the auto orientation plugin for ideas on implementing this as a plugin myself?

ianpaschal, 2018-04-11:
Yeah, if you have some coding experience I would try it. It's not hard at all from a basic implementation perspective. The two things that could make it "more trouble than it's worth" which immediately come to mind are adding a bunch of new UI elements and testing (what happens when you have several objects selected. what happens when you reset or recenter an object which has been rotated, etc.) If you're willing to take on that "responsibility" I say go for it.

fieldOfView, 2018-04-11:
I don’t want to discourage you from getting involved in Cura or plugin development, but the functionality you propose is far from trivial to implement. For one, you would need to be able to “pick” a face (to get at it’s orientation information), and Cura does not have that functionality yet.

I have implemented depth-picking for the support-blocker, and I may add face-picking for a new support-related tool.

ianpaschal, 2018-04-11:
@fieldOfView I don't think @terryspitz is talking about that. He just wants to make more buttons to do things like "rotate 90º on x axis, rotate 90º on y axis, rotate 90º on z axis" etc.

jmeeter, 2022-07-24:
Did this ever get implemented?