Ultimaker/Cura #3639: Project opened as Mesh

BagelOrb, 2018-04-09:
**Application Version**
Uhm, two different versions, e.g. Cura 3.3-beta and Cura 3.2.1

**Printer**
UM3, probably any

**Steps to Reproduce**
- [Always open as project] (or always ask)
- load .curaproject.3mf from a later version, e.g. load 
[test_0_cube.curaproject.3mf](https://github.com/Ultimaker/Cura/files/1889717/test_0_cube.curaproject.3mf.remove_extension.zip)
into Cura 3.2.1

**Actual Results**
Project loaded as Mesh

**Expected results**
Warning or error that the file is a project file but that it cannot be opened as project due to versions issues.

**Additional Information**
Might be related to https://github.com/Ultimaker/Cura/issues/3587

LipuFei, 2018-04-11:
@BagelOrb Can you try again? That seems to be fixed in the current 3.3 branch.

BagelOrb, 2018-04-11:
Uhm.... No I can't try this on 3.3, because 3.4 isn't out yet.

Maybe you misread the bug description?!

LipuFei, 2018-04-11:
You said 3.3 beta, right? That's code in the 3.3 branch. I was asking you to try that.

BagelOrb, 2018-04-11:
The bug report is about earlier versions not giving a bug when loading
projects from later versions. I can not test that on Cura 3.3, because I
don't have a project file of Cura 3.4 because it doesn't exist yet.

Please read the bug report carefully. It's about loading a project from 3.3
into 3.2.1.

Ghostkeeper, 2018-05-03:
I don't think there is much that we can do about this though. We have no guarantees of how you would recognise a Curaproject file in future versions of Cura. Currently we check for the `version` property in the `general` category of the `/cura/preferences.cfg` file, but if we renamed the `cura` folder to `project_data` or whatever, then all bets are off. We cannot make this work for future versions.

BagelOrb, 2018-05-03:
I get it if you don't want to fix this, but don't say this is impossible to fix just because you might decide to fuck it up in the future!

We should just pledge that we keep the version number at the same place for now and in the future.

I think it would be a nice lil feat if Cura could warn me that I am trying to open a file which it cannot handle correctly, but I get that this idea might be low priority.

LipuFei, 2018-07-02:
For Dev: CURA-5526

LipuFei, 2018-07-02:
After some discussion, we decided to add a warning when a older Cura tries to open a project file that's created by a newer Cura, saying that it cannot import the new project setup because it comes from the future, but it will import the models in that project file.

nallath, 2021-09-29:
CURA-5526 has been closed for some time now, so this can also be closed.

closed 2021-09-29 by nallath