Ultimaker/Cura #3640: Single layer prints cause layer view bugs

BagelOrb, 2018-04-09:
**Application Version**
Cura 3.3-beta 

**Platform**
Linux Mint AppImage (likely any)

**Display Driver**
Nvidea-384 (unlikely to be relevant)

**Printer**
UM3 (likely any)

**Steps to Reproduce**
- load any model.
- let it slice
- go to layer view
- disable uniform scaling
- scale Z down to 0.3mm (to the initial layer height)

**Actual Results**
- Layer view slider still indicates the old number of layers, instead of 1
- You see non-existing extrusion moves to the left front corner in the layer view

**Expected results**
- Layer slider indicates 1
- No visualization bugs.

Vandrasc, 2018-04-09:
Nice catch, devs see CURA-5207.

ianpaschal, 2018-07-02:
Fixed 16 May 2018.

closed 2018-07-02 by ianpaschal