Ultimaker/Cura #3643: [3.2.1] Startup Crash Lost Settings and machines

chepo92, 2018-04-10:
**Application Version**
3.2.1

**Platform**
Win7 

**Display Driver**
Intel HD 3000 

**Printer**
Not conected printer, but had left an arduino UNO (with GRBL) conected over USB

**Steps to Reproduce**
Just start cura, loading window and crashed, "problem starting cura, had two options restore defaults or close, first chose close, tried disconecting USB devices, retry... crash again, close, reboot pc... retry, crash, close, retry, chose restore defaults then all config lost, but cura started

**Actual Results**
Configurations and machine profiles lost, it said it would backup configs, still looking for backup files 

Attached log. 
[crash.txt](https://github.com/Ultimaker/Cura/files/1892535/crash.txt)

Vandrasc, 2018-04-10:
@LipuFei, Is this what we already addressed in 3.3? Thank you!

LipuFei, 2018-04-10:
Hi @chepo92 , looks like you have some corrupted data somehow. I don't know how they got there. I think if you remove the following corrupted files it should fine. If you check those files, you can find they only contain bytes `\x00`

In `machine_instances` directory:
```
MMK+HB.global.cfg
MMK+2x.global.cfg
Prusa+i3.global.cfg
Multimaker+S.global.cfg
```

In `extruders` directory:
```
custom_extruder_1+%234.extruder.cfg
custom_extruder_6+%236.extruder.cfg
custom_extruder_8+%233.extruder.cfg
Prusa+i3+fdmextruder.extruder.cfg
custom_extruder_8+%235.extruder.cfg
custom_extruder_7+%235.extruder.cfg
custom_extruder_7+%234.extruder.cfg
custom_extruder_3+%232.extruder.cfg
custom_extruder_5+%232.extruder.cfg
custom_extruder_1+%233.extruder.cfg
custom_extruder_4+%236.extruder.cfg
custom_extruder_2+%233.extruder.cfg
custom_extruder_5+%236.extruder.cfg
custom_extruder_8+%236.extruder.cfg
custom_extruder_5+%235.extruder.cfg
custom_extruder_2+%234.extruder.cfg
custom_extruder_1+%235.extruder.cfg
custom_extruder_2+%236.extruder.cfg
custom_extruder_7+%233.extruder.cfg
custom_extruder_4+%233.extruder.cfg
custom_extruder_3+%236.extruder.cfg
ultimaker_original_dual_2nd+%232.extruder.cfg
custom_extruder_2+%232.extruder.cfg
custom_extruder_3+%235.extruder.cfg
custom_extruder_5+%234.extruder.cfg
custom_extruder_1+%236.extruder.cfg
custom_extruder_6+%235.extruder.cfg
custom_extruder_2+%235.extruder.cfg
ultimaker_original_dual_1st+%232.extruder.cfg
custom_extruder_6+%234.extruder.cfg
custom_extruder_7+%232.extruder.cfg
custom_extruder_3+%234.extruder.cfg
custom_extruder_4+%232.extruder.cfg
custom_extruder_4+%235.extruder.cfg
custom_extruder_8+%234.extruder.cfg
```

Ghostkeeper, 2018-05-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/3561

closed 2018-05-01 by Ghostkeeper