Ultimaker/Cura #3644: Place Surface on Bed feature request

greghodson, 2018-04-10:
Simplify 3D has a very useful feature called "Place surface on bed". Press Ctrl+L, then click the surface you want flat on the bed...that's it. A very fast way to align even 90 degree rotates, but even more invaluable when aligning to an angle which is not known. Lay lat does a similar job, but there are more steps and I find that I can't always predict which surface it will choose to align with the bed.

I would love this feature in Cura.

ianpaschal, 2018-04-10:
I don't think we will implement this. There's already a plugin for this in the plugin browser and clicking the object, hitting a key, and selecting a surface isn't faster or easier than just clicking the object with the rotate tool and rotating 90º or 180º or whatever.

greghodson, 2018-04-10:
That's great! What's the plugin called? Is it one of these, or do I need to go elsewhere to download it? None of the ones in this list sound like they will do what I described but I will be very happy if one of them has this feature.
[image]

greghodson, 2018-04-10:
@ChrisTerBeke as mentioned in #3637 the auto-rotate feature does not seem to have anything like a "Place surface on bed" feature. Am I missing something here? 
What I am talking about is this:
https://www.youtube.com/watch?v=XJi4_ScDLtE

Ghostkeeper, 2018-05-03:
No, indeed there is no such feature.

I think that it'll become considerably easier to implement this soon, because we have one feature planned now that requires to know the normal vector of the surface you're clicking on (sorry, details are vague still so we don't want to make promises of anything yet). Once that's in place, rotating the model such that the normal vector you're clicking on is pointing down becomes trivial.

fieldOfView, 2018-05-03:
>  one feature planned now that requires to know the normal vector of the surface you're clicking on

oooh!

The technique I use to get the picked depth for the Support Blocker tool is not going to be work, because Qt can only get a 8 bit per channel RGB image from a render buffer. For a face normal you need a 16 bit float per channel/axis. I have been trying to find a way to pick a face index so you can get the actual normal of the picked face without the rendering artifacts/precision limitations of rendering a normalmap.

ianpaschal, 2018-05-04:
Yeah I did a lot of that sort of thing in the 3D drawing program I made for my graduation assignment. This was for drawing splines on 3D scans of hands so they were quite high poly and still performant in JS so I'm confident we can do it cleanly in Python as well.

greghodson, 2018-05-04:
Thanks for the update. That would be awesome. I am also very intrigued at the other feature which I think was hinted at before (support related) I can make some guesses but I am looking forward to seeing what it is :)

AbeFM, 2018-05-13:
I've had several problems getting objects on the bed in Cura. Some use the bonds of the file and ignore the actual faces.

Rotating works poorly, and the "lay flat" often leaves the part with support under all but a single point, clearly not "flat".

Putting a face on the bed would fix most of my troubles in this area.

nallath, 2018-05-14:
The auto-align tries to compute the orrientation that has the lowest amount of overhangs. It's far better than the lay flat algorithm, but it doesn't take flatness of surfaces into account.

AbeFM, 2018-05-14:
The purpose for me in "laying flat" is to increase the contact area of the part and the bed to minimize print failures and part swaying.

Three feet of a printed stool on the bed works better than one foot on the bed and two sitting on support towers, often.

Also you end up with parts with a half-layer filled on the top or bottom because the object itself is square but not sitting level on the bed.

Auto support aware orientation is awesome, but unrelated to the request for putting a surface on the bed I feel.

iasj, 2019-09-23:
> No, indeed there is no such feature. I liked the user experience of S3D there to flatten a model. It's much easier to use than Cura's automatic functionality that guesses wrong half the time.
> 
> I think that it'll become considerably easier to implement this soon, because we have one feature planned now that requires to know the normal vector of the surface you're clicking on (sorry, details are vague still so we don't want to make promises of anything yet). Once that's in place, rotating the model such that the normal vector you're clicking on is pointing down becomes trivial.
> 
> Let's wait until then.

I don't really know how Cura was written, but when I check a stl file, it has the coordinates of each triangle vertices and the coordinates of the vector normal to that face (the triangle).

fieldOfView, 2019-09-23:
@iasj, have you seen Cura 4.3 beta? It has the requested functionality.

iasj, 2019-09-23:
@fieldOfView, not really. I'm kinda afraid of updating my system now. I need to meet a deadline next week, so I'm not taking any chances.

Ghostkeeper, 2019-09-24:
Yeah, laying a triangle flat on the bed is reasonably simple. The problem was with determining which triangle was clicked on, when clicking on a certain pixel, viewing from a certain angle with occlusion etc. Luckily that should all be solved now so I'm closing this ticket. It's implemented for 4.3.