Ultimaker/Cura #3645: Combing in any mode will create travel move over holes in parts

Miranio, 2018-04-10:
**Application Version**
3.2.1

**Platform**
Windows 7

**Printer**
Anet A8 highly modified

**Steps to Reproduce**
Use part with hole in it, multiply it and prepare the print. You'll notice travel move over those holes as if cura consider them as printed area. They were printed area before right, but on the current layer they aren't so travel move should avoid them.

**Actual Results**
[image]

[image]

**Expected results**
The nozzle should stay over printed area of the actual layer being printed and not over printed area 100 layers before. The layer should avoid part also once it exited one part and enter the next one right where print should begin. Right now it's travelling in a straight line not avoiding part at all.

BagelOrb, 2018-04-12:
These prints look rather thin. Maybe they are too thin to comb on the inside.

However they should avoid the parts in any case, so that could be a bug.

So you have the Hop settings enabled?

Miranio, 2018-04-13:
Actually they are 1mm thick walls at the top section and I print with a 0.4mm nozzle, so there is enough room to comb inside

That was a behavior I noticed on other prints but those one create some issue with inside travels.

Yes z-hop is enabled.

BagelOrb, 2018-04-27:
> z-hop is 0.4mm right now.
>
> Z-hop is used to not drag the head over surface of small footprint risking
> ripping them off of the build plate.
>
> Combing would exit the part toward the outside closest to the end of print
> and would then result in stringing going out instead of in and thus are
> much much more easy to clean. Usually also when there is a hole in a model,
> it is for functional issue and thus this hole need to be with greater
> quality than the outside shell otherwise if it was only to save filament,
> the infill options of cura already provide plenty of adjustment for this
> and thus don't require modelling modification.
>
> Right now the only option for stringing on the outside is to drag the
> nozzle over the part at high speed and risk breaking them of of the build
> plate, thus why I wanted z-hop AND combing.

Yeah. Sounds reasonable.

Now that I think about it I don't really see why we automatically disabled
combing for z hops.

This should be an easy fix. It's just removing an artificial restriction,
if I remember correctly.

BagelOrb, 2018-04-30:
Come to think of it it might be not that easy.

Changing this might change the Z hop behavior when the nozzle has to print some material which is totally surrounded by material of the other extruder. The functionality of Only Z Hop When Stuck would become different.

Ghostkeeper, 2018-05-25:
We've recently made some adjustments to combing for Cura 3.4 that could make it successfully comb in thin objects like these.

> Now that I think about it I don't really see why we automatically disabled combing for z hops.

The idea was that it would no longer be necessary to z-hop if you don't cross the walls. We didn't think of the problem where a nozzle would scrape a small object off the build plate by going over infill without hop.

A second argument still holds though: If you z-hop, combing should no longer be necessary. If you're in mid-air it doesn't make sense to walk through a maze of walls which are way below your nozzle. I still hold by that. After the z-hop it should land inside your model (infill if possible), so there should be no stringing going inside either. So I hope that, since stringing going in is not a problem, a solution to your issue would be to only use z-hop.

Does it properly land in the middle of your model? Maybe if you increase the z-hop height a bit? We use 2mm by default in Ultimaker's profiles iirc.

nallath, 2019-04-04:
Since there hasn't been any response for the pas 10 months or so, I'm going to assume that the fixes in 3.4 resolved it.

closed 2019-04-04 by nallath