Ultimaker/Cura #3646: tuning head distance from bed with sensor

tmirante, 2018-04-10:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

tmirante, 2018-04-10:
i should like to write a plugin to tune head distance form bed in various points simply using inductive sensor light without automation . is there an example of head positioning plugin to start from ?

ianpaschal, 2018-04-12:
You can find existing plugins which are bundled with Cura [here](https://github.com/Ultimaker/Cura/tree/master/plugins). A lot of Cura is actually made out of plugins.

There's additional information on creating plugins [here](https://github.com/Ultimaker/Cura/wiki/Plugin-Directory).

Beyond that we can't really offer much help. You'll have to try some things yourself to get started.

closed 2018-04-12 by ianpaschal

fieldOfView, 2018-04-12:
Note that AFAIK the versions of legacy Cura that have made by Ultimaker never had a way of tuning head distance from bed with sensor, nor were they ever name version 1.x. So I think you are talking about a 3rd party version of Cura. 

I think the plugin that will be most helpful to get you started if you want to contribute such a plugin is the BedLevelMachineAction which can be found here:
https://github.com/Ultimaker/Cura/tree/master/plugins/UltimakerMachineActions

tmirante, 2018-04-12:
Ian i worked on UMA plugin but i had some troubles  so the reason of  my request. i will work more . about cura 1.15 version i am almost sure it is from ultimaker and has a command to test head on bed around on 3 points . i can deliver the right number verison if you require .

Ghostkeeper, 2018-05-03:
Ah yes, the Level Build Plate feature is still available, but only for some printers (because it depends on the location of the bed screws). Try it with an Ultimaker Original.

tmirante, 2018-05-03:
I modified plug-in creating a tunable function with parameters. I can add
to cura in future